osrf/ros2_test_cases #1438: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: FastDDS

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

ganatrask, 2024-05-01:
Hey @marcoag I tried this and was able to run all the commands. 

Can you assign this to me and close this issue? 

Thanks.

marcoag, 2024-05-01:
Can you post the outputs of the commands run for reference?

Thanks!

ganatrask, 2024-05-01:
Sure, 

here is a screenshot with all the commands. 

[image]

marcoag, 2024-05-01:
Great, thanks!

closed 2024-05-01 by marcoag