osrf/ros2_test_cases #1439: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* OS: Ubuntu Noble 24.04
* Build type: Debian

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

fujitatomoya, 2024-05-01:
@marcoag please sign me up.

fujitatomoya, 2024-05-01:
- terminal 1

```bash
root@51cdd59e1f3e:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@51cdd59e1f3e:~# ros2 run demo_nodes_cpp talker
root@51cdd59e1f3e:~# ros2 run demo_nodes_cpp add_two_ints_server
root@51cdd59e1f3e:~# ros2 run demo_nodes_cpp parameter_blackboard
root@51cdd59e1f3e:~# ros2 run action_tutorials_cpp fibonacci_action_server
```

- terminal 2

```bash
root@51cdd59e1f3e:~# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@51cdd59e1f3e:~# ros2 node list
/talker
root@51cdd59e1f3e:~# ros2 topic list
/chatter
/parameter_events
/rosout
root@51cdd59e1f3e:~# ros2 service list
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/get_type_description
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically
root@51cdd59e1f3e:~# ros2 param list
/parameter_blackboard:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
root@51cdd59e1f3e:~# ros2 action list
/fibonacci
```

fujitatomoya, 2024-05-01:
endpoint discovery can work without problem, this can be closed.

marcoag, 2024-05-02:
Thanks for testing!

closed 2024-05-02 by marcoag