osrf/ros2_test_cases #1441: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* Build type: Debian
* DDS: FastDDS
* Chip set: ARM64
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

slowrunner, 2024-05-02:
Using Jazzy/Noble nodes running in a VM on my Mac, the Raspberry Pi5 Arm64 Noble, Jazzy Debian install, FastDDS "sees" the networked ROS 2 nodes and the listed tests all succeeded as well as other tests performed.

clalancette, 2024-05-02:
Fantastic, thank you for testing!

closed 2024-05-02 by clalancette