osrf/ros2_test_cases #1442: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* Chip set: ARM64
* OS: Ubuntu Noble 24.04
* DDS: CycloneDDS
* Build type: Debian

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

wep21, 2024-05-03:
```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.7.9-200.fc39.aarch64-aarch64-with-glibc2.39
release          : 6.7.9-200.fc39.aarch64
processor        : aarch64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

```
root@b86ef71be53f:/# ros2 run demo_nodes_cpp talker
[INFO] [1714721183.424019557] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714721184.424172513] [talker]: Publishing: 'Hello World: 2'
[INFO] [1714721185.426733637] [talker]: Publishing: 'Hello World: 3'
```
```
root@b86ef71be53f:/# ros2 run demo_nodes_cpp add_two_ints_server
```
before stop
```
root@b86ef71be53f:/# ros2 node list
/add_two_ints_server
/talker
```
after stop
```
root@b86ef71be53f:/# ros2 node list
```
before stop
```
root@b86ef71be53f:/# ros2 topic list
/chatter
/parameter_events
/rosout
```
after stop
```
root@b86ef71be53f:/# ros2 topic list
/parameter_events
/rosout
```
before stop
```
root@b86ef71be53f:/# ros2 service list
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/get_type_description
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically
/talker/describe_parameters
/talker/get_parameter_types
/talker/get_parameters
/talker/get_type_description
/talker/list_parameters
/talker/set_parameters
/talker/set_parameters_atomically
```
after stop
```
root@b86ef71be53f:/# ros2 service list
```
before stop
```
root@b86ef71be53f:/# ros2 param list
/add_two_ints_server:
  one_shot
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
/talker:
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```
after stop
```
root@b86ef71be53f:/# ros2 param list
```

marcoag, 2024-05-03:
Looking good, thanks!

closed 2024-05-03 by marcoag