osrf/ros2_test_cases #1444: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* Build type: Binary
* DDS: FastDDS
* Chip set: AMD64
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

Kartik9250, 2024-05-03:
I would like to work on this.

samuel-ongzx, 2024-05-15:
## Setup
OS: Docker: `ghcr.io/sloretz/ros-testing:jazzy-desktop`
Chip set: AMD64
Build type: Binary (downloaded from release page)
DDS: FastDDS

## Process 
- [x] 1. Node Discovery `ok`
- [x] 2. Topic Discovery `ok`
- [x] 3. Service Discovery `ok`
- [x] 4. Parameter Discovery `ok`

## Verification
Node ran: `RMW_IMPLEMENTATION==rmw_fastrtps_cpp ros2 run demo_nodes_cpp talker`
[image]

Output of commands:
[image]

closed 2024-05-15 by ahcorde