osrf/ros2_test_cases #1447: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* Build type: Binary
* Chip set: ARM64
* DDS: FastDDS
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

Tacha-S, 2024-05-03:
Looks good to me.

<details>
<summary>Step 1 - Node discovery</summary>

```bash

# Terminal 1
ubuntu@8ab521739af6:/$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-05-03-07-29-27-038277-8ab521739af6-1113
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [1116]
[INFO] [robot_state_publisher-2]: process started with pid [1117]
[INFO] [dummy_joint_states-3]: process started with pid [1118]
[INFO] [dummy_laser-4]: process started with pid [1119]
[dummy_laser-4] [INFO] [1714721367.100502715] [dummy_laser]: angle inc: 0.004363
[dummy_laser-4] [INFO] [1714721367.100557194] [dummy_laser]: scan size: 1081
[dummy_laser-4] [INFO] [1714721367.100561795] [dummy_laser]: scan time increment:       0.000000
[robot_state_publisher-2] [INFO] [1714721367.104251797] [robot_state_publisher]: Robot initialized

# Terminal 2
ubuntu@8ab521739af6:/$ ros2 node list
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher

# After stopping dummy_robot_bringup
ubuntu@8ab521739af6:/$ ros2 node list
ubuntu@8ab521739af6:/$
```

</details>

<details>
<summary>Step 2 - Topic discovery</summary>

```bash

# Terminal 1
ubuntu@8ab521739af6:/$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-05-03-07-29-27-038277-8ab521739af6-1113
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [1116]
[INFO] [robot_state_publisher-2]: process started with pid [1117]
[INFO] [dummy_joint_states-3]: process started with pid [1118]
[INFO] [dummy_laser-4]: process started with pid [1119]
[dummy_laser-4] [INFO] [1714721367.100502715] [dummy_laser]: angle inc: 0.004363
[dummy_laser-4] [INFO] [1714721367.100557194] [dummy_laser]: scan size: 1081
[dummy_laser-4] [INFO] [1714721367.100561795] [dummy_laser]: scan time increment:       0.000000
[robot_state_publisher-2] [INFO] [1714721367.104251797] [robot_state_publisher]: Robot initialized

# Terminal 2
ubuntu@8ab521739af6:/$ ros2 topic list
/joint_states
/map
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static

# After stopping dummy_robot_bringup
ubuntu@8ab521739af6:/$ ros2 topic list
/parameter_events
/rosout
```

</details>
[... 101 lines cut ...]
  publish_frequency
  qos_overrides./joint_states.subscription.depth
  qos_overrides./joint_states.subscription.history
  qos_overrides./joint_states.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  qos_overrides./tf_static.publisher.depth
  qos_overrides./tf_static.publisher.history
  qos_overrides./tf_static.publisher.reliability
  robot_description
  start_type_description_service
  use_sim_time

# After stopping dummy_robot_bringup
ubuntu@8ab521739af6:/$ ros2 param list
ubuntu@8ab521739af6:/$
```

</details>

closed 2024-05-03 by marcoag

knmcguire, 2024-05-03:
Hi! I noticed that both the dummy bringup and the ros2 node list seem to have been done on the same machine as it has the same name: `ubuntu@8ab521739af6`.

Are you sure these test have done correctly? I've been getting issues with the tests for the cyclonedds variant of this

marcoag, 2024-05-03:
@knmcguire interesting, what issues are you facing?

knmcguire, 2024-05-03:
Let's discuss my network issues in this issue https://github.com/osrf/ros2_test_cases/issues/1448

For this particular issue, from the steps that @Tacha-S has done it seems like it has been done on the same machine (at least, the name is the same).

Tacha-S, 2024-05-04:
I had two dockers running and accessing the same container. It's my mistake.

I tested it again and it seems to work fine with FastDDS.

<details>
<summary>Step 1 - Node discovery</summary>

```bash

# Terminal 1
ubuntu@a86521a0674e:/$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-05-04-00-58-36-733398-a86521a0674e-104
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [109]
[INFO] [robot_state_publisher-2]: process started with pid [110]
[INFO] [dummy_joint_states-3]: process started with pid [111]
[INFO] [dummy_laser-4]: process started with pid [112]
[dummy_laser-4] [INFO] [1714784316.945798363] [dummy_laser]: angle inc: 0.004363
[dummy_laser-4] [INFO] [1714784316.945907880] [dummy_laser]: scan size: 1081
[dummy_laser-4] [INFO] [1714784316.945918535] [dummy_laser]: scan time increment:       0.000000
[robot_state_publisher-2] [INFO] [1714784316.954184998] [robot_state_publisher]: Robot initialized

# Terminal 2
ubuntu@bc16b8309314:/$ ros2 node list
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher

# After stopping dummy_robot_bringup
ubuntu@bc16b8309314:/$ ros2 node list
ubuntu@bc16b8309314:/$
```

</details>

<details>
<summary>Step 2 - Topic discovery</summary>

```bash

# Terminal 1
ubuntu@a86521a0674e:/$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-05-04-01-02-37-266788-a86521a0674e-169
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [172]
[INFO] [robot_state_publisher-2]: process started with pid [173]
[INFO] [dummy_joint_states-3]: process started with pid [174]
[INFO] [dummy_laser-4]: process started with pid [175]
[dummy_laser-4] [INFO] [1714784557.328558093] [dummy_laser]: angle inc: 0.004363
[dummy_laser-4] [INFO] [1714784557.328601387] [dummy_laser]: scan size: 1081
[dummy_laser-4] [INFO] [1714784557.328605261] [dummy_laser]: scan time increment:       0.000000
[robot_state_publisher-2] [INFO] [1714784557.332875613] [robot_state_publisher]: Robot initialized

# Terminal 2
ubuntu@bc16b8309314:/$ ros2 topic list
/joint_states
/map
/parameter_events
/robot_description
/rosout
/scan
/tf
/tf_static

# After stopping dummy_robot_bringup
ubuntu@bc16b8309314:/$ ros2 topic list
/parameter_events
/rosout
```
[... 103 lines cut ...]
  publish_frequency
  qos_overrides./joint_states.subscription.depth
  qos_overrides./joint_states.subscription.history
  qos_overrides./joint_states.subscription.reliability
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  qos_overrides./tf.publisher.depth
  qos_overrides./tf.publisher.durability
  qos_overrides./tf.publisher.history
  qos_overrides./tf.publisher.reliability
  qos_overrides./tf_static.publisher.depth
  qos_overrides./tf_static.publisher.history
  qos_overrides./tf_static.publisher.reliability
  robot_description
  start_type_description_service
  use_sim_time

# After stopping dummy_robot_bringup
ubuntu@bc16b8309314:/$ ros2 param list
ubuntu@bc16b8309314:/$
```

</details>