osrf/ros2_test_cases #1448: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* OS: Ubuntu Noble 24.04
* Build type: Binary
* Chip set: ARM64
* DDS: CycloneDDS

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

knmcguire, 2024-05-03:
I'll be working on this with the dummy robot launch from another machine (AMD64 with jazzy docker) and discovery from a raspberry pi 5 with Ubuntu Noble and the binary install of ROS2 with the cyclonedds removed from the skip tags

knmcguire, 2024-05-03:
Terminal 1 (jazzy docker on AMD64):
```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```

terminal 2 (Raspberri pi 5 ubuntu nobel)
```
$ . ~/ros2_jazzy/ros2-linux/setup.bash
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 node list
```

Nothing shown...

On the same machine it does seem to work so I'm currently investigating what is going wrong, especially since this issue does seem to work: https://github.com/osrf/ros2_test_cases/issues/1447.

knmcguire, 2024-05-03:
So I am able to ping my raspberry pi but the docker container on my AMD64 machine can not be pinged by my raspberry pi. I guess that it won't be possible to do that... Perhaps I should assign a port to my docker container then.

knmcguire, 2024-05-03:
So if I start the jazzy docker with port forwarding I am able to ping with adding `-p 80`

```
docker run -p 8080:80 --rm=true -ti ghcr.io/sloretz/ros-testing:jazzy-desktop bash
```
But... still no discovery. Think that I'm hitting an edge case here.

I'll try to acquire a native install of ubuntu Noble on another machine to not hit this issue, but if anyone else managed to test this, go for it

knmcguire, 2024-05-06:
I saw in the other issue that someone started 2 docker instances on the same machine. Is that also considered over network? It would make things easier for sure but not sure if we'd be testing the right thing.

knmcguire, 2024-05-06:
Got it to work with the docker image by adding `--network=host` to the jazzy docker startup. 

I ran the docker for jazzy on an ubuntu 22.04 which was sharing the same wifi as the raspberry pi 5

- [ ] Step 1
```
$ ros2 node list
/dummy_joint_states
/dummy_laser
/dummy_map_server
/robot_state_publisher
```

- [ ] step 2

```
$ ros2 topic list
/joint_states
/map
...
/tf
/tf_static
```

- [ ] step 3

```
$ ros2 service list/dummy_joint_states/describe_parameters
/dummy_joint_states/get_parameter_types
/dummy_joint_states/get_parameters
...
/robot_state_publisher/set_parameters
/robot_state_publisher/set_parameters_atomically
```

- [ ] step 4

```
$ ros2 param list
/dummy_joint_states:
  qos_overrides./parameter_events.publisher.depth
...
  start_type_description_service
  use_sim_time
```

**It works!**

marcoag, 2024-05-07:
Great! Thanks for testing!

closed 2024-05-07 by marcoag