osrf/ros2_test_cases #1450: Rviz

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: FastDDS

## Links

* [Repo URL](https://github.com/ros2/rviz)

## Process

    
            
### Check Rviz
            
        

    
Launch using the `rviz2` command.

The rviz window should be generally usable and should be able to load some of the discovered visualizations.

Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rviz2
```
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
                
    

    
        
#### Expected Results
        
        
You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
                
You should see a laser scan emitting from the waving end of the dummy robot.
                
Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

ganatrask, 2024-05-01:
Hey @marcoag , I tried to follow the instruction for this issue, I was able to run the launch file but had some issue with running RVIZ2

`[rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre_vendor-prefix/src/ogre_vendor/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 246)
[ERROR] [1714601313.321657126] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre_vendor-prefix/src/ogre_vendor/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 246)
[ERROR] [1714601313.321699050] [rviz2]: Unable to create the rendering window after 100 tries
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create the rendering window after 100 tries
Aborted (core dumped)
`

Can you please look into this?

wjwwood, 2024-05-02:
In my experience this occurs due to issues with video card drivers in a vm. Can you try the glxgears command? Does that run?

ganatrask, 2024-05-02:
This could be an issue, but usually I run ROS in VM never faced any issue.

Also tried to `glxgears` it is running as it should.

clalancette, 2024-05-02:
Can you give us some more details about your installation?  In particular, what is your host platform and what virtual machine technology are you using?

ganatrask, 2024-05-02:
Sure, I'm using Ubuntu 24.04 on a VirtualBox VM on Ubuntu 22.04 as my host system.

slowrunner, 2024-05-03:
I performed this configuration test 
OS: Linux Ubuntu-Noble-FusionVM 6.8.0-31-generic #31-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 00:40:06 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

Chip set: AMD64  8 cores 8GB memory

Build type: Debian

DDS: FastDDS

Steady 31 fps - camera moves, zooms no problems - only two "dropping message ... timestamp earlier than all data in cache"  (normal to see some of these INFO).

MGupta28, 2024-05-03:
I was able to get the required result in the docker image and run the rviz

Output for terminal 1:

[image]

Output for terminal 2:

[image]

Rviz2 window:

[image]

The result in rviz window is as expected
- dummy robot's kinematic tree waving back and forth in the rviz window.
- laser scan emitting from the waving end of the dummy robot.

clalancette, 2024-05-06:
All right, looks like people were eventually successful here.  I'm going to go ahead and close this out.

Thanks for testing!

closed 2024-05-06 by clalancette