osrf/ros2_test_cases #1451: Rviz

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* OS: Ubuntu Noble 24.04
* Build type: Debian

## Links

* [Repo URL](https://github.com/ros2/rviz)

## Process

    
            
### Check Rviz
            
        

    
Launch using the `rviz2` command.

The rviz window should be generally usable and should be able to load some of the discovered visualizations.

Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rviz2
```
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
                
    

    
        
#### Expected Results
        
        
You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
                
You should see a laser scan emitting from the waving end of the dummy robot.
                
Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

hwoithe, 2024-05-02:
I followed the instructions above and everything worked as expected.

[image]

The dummy_robot_bringup nodes are using CycloneDDS:
```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
$ lsof -P -p `pidof dummy_laser` 2>/dev/null | grep cyclonedds
dummy_las 819768 hwoithe  mem    REG     0,60   473584 22175331 /opt/ros/jazzy/lib/librmw_cyclonedds_cpp.so
$ lsof -P -p `pidof dummy_joint_states` 2>/dev/null | grep cyclonedds
dummy_joi 819767 hwoithe  mem    REG     0,60   473584 22175331 /opt/ros/jazzy/lib/librmw_cyclonedds_cpp.so
$ lsof -P -p `pidof robot_state_publisher` 2>/dev/null | grep cyclonedds
robot_sta 819766 hwoithe  mem    REG     0,60   473584 22175331 /opt/ros/jazzy/lib/librmw_cyclonedds_cpp.so
$ lsof -P -p `pidof dummy_map_server` 2>/dev/null | grep cyclonedds
dummy_map 819765 hwoithe  mem    REG     0,60   473584 22175331 /opt/ros/jazzy/lib/librmw_cyclonedds_cpp.so
```

RViz is using CycloneDDS:
```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ rviz2
$ lsof -P -p `pidof rviz2` 2>/dev/null | grep cyclonedds
rviz2   819831 hwoithe  mem       REG               0,60   473584 22175331 /opt/ros/jazzy/lib/librmw_cyclonedds_cpp.s

```

Here is my system's information:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 24.04 LTS
Release:    24.04
Codename:   noble

$ glxinfo | grep -i renderer
OpenGL renderer string: NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2
```

clalancette, 2024-05-02:
Thanks for testing!

closed 2024-05-02 by clalancette