osrf/ros2_test_cases #1452: Rviz

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Chip set: AMD64
* DDS: ConnextDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros2/rviz)

## Process

    
            
### Check Rviz
            
        

    
Launch using the `rviz2` command.

The rviz window should be generally usable and should be able to load some of the discovered visualizations.

Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rviz2
```
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
                
    

    
        
#### Expected Results
        
        
You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
                
You should see a laser scan emitting from the waving end of the dummy robot.
                
Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

Kartik9250, 2024-05-06:
I would like to work on this

Kartik9250, 2024-05-06:
Hey it works fine,. But as i pointed before ubuntu nobel has the wayland issue. So i had to  run this ciommand `export QT_QPA_PLATFORM=xcb `
[image]

clalancette, 2024-05-06:
> Hey it works fine,. But as i pointed before ubuntu nobel has the wayland issue. So i had to run this ciommand `export QT_QPA_PLATFORM=xcb `

Right, thanks for the heads up.  I've opened up https://github.com/ros2/ros2_documentation/pull/4375 to add this to our troubleshooting guide.

With that, I'm going to go ahead and close this.  Thanks for testing!

closed 2024-05-06 by clalancette