osrf/ros2_test_cases #1454: Rviz

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Chip set: ARM64
* OS: Ubuntu Noble 24.04
* DDS: CycloneDDS
* Build type: Debian

## Links

* [Repo URL](https://github.com/ros2/rviz)

## Process

    
            
### Check Rviz
            
        

    
Launch using the `rviz2` command.

The rviz window should be generally usable and should be able to load some of the discovered visualizations.

Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rviz2
```
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
                
    

    
        
#### Expected Results
        
        
You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
                
You should see a laser scan emitting from the waving end of the dummy robot.
                
Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

knmcguire, 2024-05-03:
I'll be testing this issue now on a raspberry pi 5- Ubuntu Noble- Jazzy and Cyclonedds installed by debian packages

knmcguire, 2024-05-03:
Terminal 1:
```
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/bitcraze/.ros/log/2024-05-03-11-54-59-504918-bitcraze-RP5-4036
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [4039]
[INFO] [robot_state_publisher-2]: process started with pid [4040]
[INFO] [dummy_joint_states-3]: process started with pid [4041]
[INFO] [dummy_laser-4]: process started with pid [4042]
[dummy_laser-4] [INFO] [1714730099.765424535] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1714730099.765546997] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1714730099.765568423] [dummy_laser]: scan time increment: 	0.000000
[robot_state_publisher-2] [INFO] [1714730099.823362939] [robot_state_publisher]: Robot initialized

```
Terminal 2:
```
$ rviz2
QSocketNotifier: Can only be used with threads started with QThread
[INFO] [1714730141.758659580] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1714730141.758813709] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[ERROR] [1714730141.797879232] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-aarch64-linux-gnu/ogre_vendor-prefix/src/ogre_vendor/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 246)
...
[ERROR] [1714730141.808124283] [rviz2]: Unable to create the rendering window after 100 tries
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create the rendering window after 100 tries
Aborted (core dumped)
```

**Seems not okay!** Rviz2 never starts up. 

However, when I try to do this with the default RMW, I get the same error so I don't think this is Cyclone DDS related. I'll try to figure out what went wrong as this test did pass for someone else https://github.com/osrf/ros2_test_cases/issues/1453

knmcguire, 2024-05-03:
Managed to get rviz2 to work via the [fix proposed here](https://github.com/ros-visualization/rviz/issues/1442#issuecomment-553900795).  I had export this environment variable first: `QT_QPA_PLATFORM=xcb `

Terminal 1:

```
bitcraze@bitcraze-RP5:~$ source /opt/ros/jazzy/setup.bash
bitcraze@bitcraze-RP5:~$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
bitcraze@bitcraze-RP5:~$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/bitcraze/.ros/log/2024-05-03-12-12-16-779108-bitcraze-RP5-5209
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [5212]
[INFO] [robot_state_publisher-2]: process started with pid [5213]
[INFO] [dummy_joint_states-3]: process started with pid [5214]
[INFO] [dummy_laser-4]: process started with pid [5215]
[robot_state_publisher-2] [INFO] [1714731136.974925894] [robot_state_publisher]: Robot initialized
[dummy_laser-4] [INFO] [1714731137.011838599] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1714731137.012032063] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1714731137.012079527] [dummy_laser]: scan time increment: 	0.000000
```

Terminal 2

```
$ source /opt/ros/jazzy/setup.bash
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ QT_QPA_PLATFORM=xcb rviz2
[INFO] [1714731155.262512182] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1714731155.262638516] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[INFO] [1714731155.347845636] [rviz2]: Stereo is NOT SUPPORTED
```

knmcguire, 2024-05-03:
With screenshot

[image]

clalancette, 2024-05-06:
Fantastic, thank you for testing!

I think what we should do is probably put a note in the Troubleshooting section of our documentation about adding in your workaround, as it seems like that will be needed by anyone using Wayland natively.  I'll go ahead and open a PR for that.

clalancette, 2024-05-06:
All right, I've opened up https://github.com/ros2/ros2_documentation/pull/4375 to deal with this problem, so I'm going to go ahead and close this.

Thanks for testing!

closed 2024-05-06 by clalancette

knmcguire, 2024-05-06:
That seems like the solution for now for this. I'm still impressed on what the RP5 can do with this :)