osrf/ros2_test_cases #1456: Rviz

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Build type: Binary
* DDS: FastDDS
* Chip set: AMD64
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros2/rviz)

## Process

    
            
### Check Rviz
            
        

    
Launch using the `rviz2` command.

The rviz window should be generally usable and should be able to load some of the discovered visualizations.

Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rviz2
```
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
                
    

    
        
#### Expected Results
        
        
You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
                
You should see a laser scan emitting from the waving end of the dummy robot.
                
Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

Kartik9250, 2024-05-03:
I would like to work on this

Kartik9250, 2024-05-03:
Rviz cannot be open, therefore wasn't able to test it. I even tried running it via the `ros2 run rviz2 rviz2`. Also, the rviz is not shown in the ubuntu applications like previous releases.
The terminal prompt the following error message:

``` QSocketNotifier: Can only be used with threads started with QThread
[INFO] [1714738480.402766954] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1714738480.402957692] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[ERROR] [1714738480.448101867] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre_vendor-prefix/src/ogre_vendor/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 246)
[ERROR] [1714738480.482488998] [rviz2]: Unable to create the rendering window after 100 tries
terminate called after throwing an instance of 'std::runtime_error'
  what():  Unable to create the rendering window after 100 tries
[1]    25305 IOT instruction (core dumped)  rviz2
```

The `[Error]` line repeats about 100 times

MGupta28, 2024-05-03:
I would like to work on this issue

MGupta28, 2024-05-03:
Output for terminal 1:

[image]

Output for terminal 2:

[image]

Rviz2 window:

[image]

The result in rviz window is as expected
- dummy robot's kinematic tree waving back and forth in the rviz window.
- laser scan emitting from the waving end of the dummy robot.

Kartik9250, 2024-05-04:
As mentioned in this ros-visualization/rviz#1442 `QT_QPA_PLATFORM=xcb`, exporting this env variable helped fixed the issue.
[image]

closed 2024-05-06 by marcoag