osrf/ros2_test_cases #1457: Rviz

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* OS: Ubuntu Noble 24.04
* DDS: CycloneDDS
* Build type: Binary
* Chip set: AMD64

## Links

* [Repo URL](https://github.com/ros2/rviz)

## Process

    
            
### Check Rviz
            
        

    
Launch using the `rviz2` command.

The rviz window should be generally usable and should be able to load some of the discovered visualizations.

Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rviz2
```
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
                
    

    
        
#### Expected Results
        
        
You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
                
You should see a laser scan emitting from the waving end of the dummy robot.
                
Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

samuel-ongzx, 2024-05-15:
## Setup
- Docker: `ghcr.io/sloretz/ros-testing:jazzy-desktop`
- DDS: CycloneDDS installed with `debian apt`
- Build type: Binary downloaded from [release page](https://github.com/ros2/ros2/releases)
- Chip set: AMD64

## Process
- [ ] Check Rviz (`ok`, node runs, rviz2 runs)
- [ ] Adding topics & visualization (`ok` world frame ok, laser can ok)
- [ ] Expected Results (`ok` see image below)

## Verification
Command ran in binary folder `source setup.bash && export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`
[image]

marcoag, 2024-05-15:
Thanks for the tests!

closed 2024-05-15 by marcoag