osrf/ros2_test_cases #1460: Rviz

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* OS: Ubuntu Noble 24.04
* Build type: Binary
* Chip set: ARM64
* DDS: CycloneDDS

## Links

* [Repo URL](https://github.com/ros2/rviz)

## Process

    
            
### Check Rviz
            
        

    
Launch using the `rviz2` command.

The rviz window should be generally usable and should be able to load some of the discovered visualizations.

Loading visualizations into the tree and using the mouse to navigate through the virtual workspace is sufficient for this test.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rviz2
```
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'TF' and click 'OK'. Under 'Displays', in the 'Global Options' section, change the 'Fixed Frame' to '/world'.
                
In RViz, in the left pane, click 'Add'. A window will pop up. In that window, under 'rviz_default_plugins', select 'LaserScan' and click 'OK'. Under 'Displays', in the 'LaserScan' section, change the 'Topic' to '/scan'.
                
    

    
        
#### Expected Results
        
        
You should see the dummy robot's kinematic tree waving back and forth in the rviz window.
                
You should see a laser scan emitting from the waving end of the dummy robot.
                
Play around, move the camera, enable and disable things in the 'Displays' pane. RViz should be stable.

knmcguire, 2024-05-03:
Looking into this now on a raspberry pi 5, ubuntu Noble with ROS2 installed by binary with connectdds removed from the skip tag in rosdep

knmcguire, 2024-05-03:
Terminal 1: 
```
$ . ~/r$ . ~/ros2_jazzy/ros2-linux/setup.bash
~$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
~$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below ----/.ros/log/2024-05-03-14-35-11-015363-bitcraze-RP5-56454
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [56457]
[INFO] [robot_state_publisher-2]: process started with pid [56458]
[INFO] [dummy_joint_states-3]: process started with pid [56459]
[INFO] [dummy_laser-4]: process started with pid [56460]
[dummy_laser-4] [INFO] [1714739711.172507532] [dummy_laser]: angle inc:	0.004363
[robot_state_publisher-2] [INFO] [1714739711.173290999] [robot_state_publisher]: Robot initialized
[dummy_laser-4] [INFO] [1714739711.174236005] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1714739711.174312135] [dummy_laser]: scan time increment: 	0.000000os2_jazzy/ros2-linux/setup.bash
bitcraze@bitcraze-RP5:~$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
bitcraze@bitcraze-RP5:~$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/bitcraze/.ros/log/2024-05-03-14-35-11-015363-bitcraze-RP5-56454
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [56457]
[INFO] [robot_state_publisher-2]: process started with pid [56458]
[INFO] [dummy_joint_states-3]: process started with pid [56459]
[INFO] [dummy_laser-4]: process started with pid [56460]
[dummy_laser-4] [INFO] [1714739711.172507532] [dummy_laser]: angle inc:	0.004363
[robot_state_publisher-2] [INFO] [1714739711.173290999] [robot_state_publisher]: Robot initialized
[dummy_laser-4] [INFO] [1714739711.174236005] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1714739711.174312135] [dummy_laser]: scan time increment: 	0.000000
```

Terminal 2: 

```
$ . ~/ros2_jazzy/ros2-linux/setup.bash
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ QT_QPA_PLATFORM=xcb rviz2
[INFO] [1714739784.584746669] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1714739784.584882540] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[INFO] [1714739784.632885594] [rviz2]: Stereo is NOT SUPPORTED
```
Printscreen and all the tests seem to work fine. Also I could rotate with it with no issues:

[image]

**Seems to work!**

Note that I had to do ` $ QT_QPA_PLATFORM=xcb rviz2` like in issue https://github.com/osrf/ros2_test_cases/issues/1454

clalancette, 2024-05-06:
Thanks for testing!

closed 2024-05-06 by clalancette