osrf/ros2_test_cases #1462: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: FastDDS

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

ganatrask, 2024-05-01:
Hello @marcoag I tested this with all the conditions mentioned above, I was able to get the expected results.

can you assign this to me and close? 

Thanks.

marcoag, 2024-05-01:
Great! Thanks for testing!

closed 2024-05-01 by marcoag

ganatrask, 2024-05-01:
Here are some sceernshots of RQT working smoothly

[image]
[image]
[image]
[image]