osrf/ros2_test_cases #1464: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Chip set: AMD64
* DDS: ConnextDDS
* Build type: Debian
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

Kartik9250, 2024-05-03:
I would like to work on this

Kartik9250, 2024-05-03:
@marcoag for `connext_dds` ussage, I would have to install the free trail from their website, right? as exporting  `RMW_IMPLEMENTATION=rmw_connextdds` gives out  this error:

```[ERROR] [1714732268.713522462] [rcl]: Error getting RMW implementation identifier / RMW implementation not installed (expected identifier of 'rmw_connextdds'), with error message 'failed to load shared library 'librmw_connextdds.so' due to dlopen error: librmw_connextdds.so: cannot open shared object file: No such file or directory, at ./src/shared_library.c:99, at ./src/functions.cpp:65', exiting with 1., at ./src/rcl/rmw_implementation_identifier_check.c:143```

marcoag, 2024-05-03:
Yes you will need some form of `ConnextDDS`, with either university, purchased or evaluation license: https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-University-Eval.html

Kartik9250, 2024-05-03:
I applied for the license, awaiting their response now.

marcoag, 2024-06-26:
Closing this once since the tutorial party is over. Thanks for the work here @Kartik9250.

closed 2024-06-26 by marcoag