osrf/ros2_test_cases #1465: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Build type: Debian
* DDS: FastDDS
* Chip set: ARM64
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

pradyum, 2024-05-02:
## Test Environment

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-31-generic-aarch64-with-glibc2.39
release          : 6.8.0-31-generic
processor        : aarch64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```

## Error

When I tried to use bag plugin "Plugins > Logging > Bag" there are some errors when opening, recording and playing the saved file. The recorded file is saved in the directory with a mcap and yaml file.

```
QSocketNotifier: Can only be used with threads started with QThread
MESA: error: ZINK: failed to choose pdev
libEGL warning: egl: failed to create dri2 screen
QLayout::removeWidget: Cannot remove a null widget.
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag_plugins/image_plugin.py", line 35, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag_plugins/image_timeline_renderer.py", line 44, in <module>
    from PIL.ImageQt import ImageQt
ImportError: cannot import name 'ImageQt' from 'PIL.ImageQt' (/usr/lib/python3/dist-packages/PIL/ImageQt.py). Did you mean: 'Image'?

rqt_bag.TimelineFrame.load_plugins() failed to load plugin "rqt_bag_plugins/BagImagePlugin":
cannot import name 'ImageQt' from 'PIL.ImageQt' (/usr/lib/python3/dist-packages/PIL/ImageQt.py)
[INFO] [1714608654.499814204] [rqt_bag.BagWidget]: Recording to /home/pc/Desktop/2024-05-02-05-40-50.
Error opening bag for recording [/home/pc/Desktop/2024-05-02-05-40-50]: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. rosbag2_py._storage.TopicMetadata(id: int, name: str, type: str, serialization_format: str, offered_qos_profiles: List[rosbag2_py._storage.QoS] = [], type_description_hash: str = '')

Invoked with: kwargs: name='/joint_states', type='sensor_msgs/msg/JointState', serialization_format='cdr', offered_qos_profiles='- history: 3\n  depth: 0\n  reliability: 1\n  durability: 2\n  lifespan:\n    sec: 9223372036\n    nsec: 854775807\n  deadline:\n    sec: 9223372036\n    nsec: 854775807\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 9223372036\n    nsec: 854775807\n  avoid_ros_namespace_conventions: false\n'
[WARN] [1714608685.032708938] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index
[WARN] [1714608685.032890395] [rosbag2_cpp]: No topics were listed in metadata.
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/time.py", line 87, in __add__
    return Time(
           ^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/time.py", line 57, in __init__
    raise OverflowError(
OverflowError: Total nanoseconds value is too large to store in C time point.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/bag_widget.py", line 281, in _handle_load_clicked
    self.load_bag(filename)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/bag_widget.py", line 318, in load_bag
    self._timeline.add_bag(bag)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/bag_timeline.py", line 175, in add_bag
    self._timeline_frame.reset_timeline()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/timeline_frame.py", line 1042, in reset_timeline
    self.reset_zoom()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/timeline_frame.py", line 1075, in reset_zoom
    end_stamp = start_stamp + Duration(seconds=5)
                ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/time.py", line 91, in __add__
    raise OverflowError('Addition leads to overflow in C storage.') from e
OverflowError: Addition leads to overflow in C storage.
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/timeline_frame.py", line 314, in paint
    self._layout()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/timeline_frame.py", line 416, in _layout
    new_history_bottom = max([y + h for (_, y, _, h) in self._history_bounds.values()]) - 1
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: max() iterable argument is empty
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/timeline_frame.py", line 314, in paint
    self._layout()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag/timeline_frame.py", line 416, in _layout
    new_history_bottom = max([y + h for (_, y, _, h) in self._history_bounds.values()]) - 1
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: max() iterable argument is empty
```

## Result

Except the Bag plugin error, All the mentioned plugins in RQT are working as expected.

slowrunner, 2024-05-02:
On TEST PLATFORM - Raspberry Pi5 / Ubuntu 23.04
```
  PLATFORM INFORMATION
Linux Pi5Noble 6.8.0-1004-raspi #4-Ubuntu SMP PREEMPT_DYNAMIC Sat Apr 20 02:29:55 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
```

Under the default signin (wm=Wayland) RQT presents two errors, but functions as expected:
- rqt invocation: QSocketNotifier: Can only be used with threads started with QThread
- plugins->visualization->plot:  QLayout::removeWidget: Cannot remove a null widget

Choosing the "settings gear" at password entry and login ubuntu on Xorg uses X11 window manager, presents one error but functions as expected:
- plugins->visualization->plot:  QLayout::removeWidget: Cannot remove a null widget

marcoag, 2024-06-26:
The issue with the timeline renderer got fixed and backported in https://github.com/ros-visualization/rqt_bag/commit/fdc34b04990257a1259e26aac766a2e04ea3d70f. Thanks for the help here @slowrunner and @pradyum!

closed 2024-06-26 by marcoag