osrf/ros2_test_cases #1468: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Build type: Binary
* DDS: FastDDS
* Chip set: AMD64
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

Kartik9250, 2024-05-03:
i would like to work on this

Kartik9250, 2024-05-03:
I found the following issues with rqt:

1. Vizualization:
    The rqt window goes lagging and goes unresponsive, by pressing wait it continues but still lags.
    [image]
    
2. Bag file Recording:
    - The  terminal prompts some errors but it is able to save bag file. Though they are empty
        [image]
        [image]
        [image]
    
    - Also, the ui doesn't has a stop or pause button, or it might be disabled due to those errors though I am not sure. 
        [image]

3. Topic monitor:
        It works just fine.
             [image]

One more thing that I would like to add, I found resizing the window a bit laggy.

samuel-ongzx, 2024-05-14:
Thanks @Kartik9250. Found similar result, except that `rqt` was not laggy on resizing. 

Setup: 
- Host Computer: `AMD64`
- Docker Container: `24.04` from `ghcr.io/sloretz/ros-testing:jazzy-desktop`
- Built Type `Binary` from [Release Page](https://github.com/ros2/ros2/releases)
- FastDDS with `apt install ros-jazzy-rmw-fastrtps-cpp`

Additional Steps needed:
- Install `python3-pyqt6`

Test Result:
- [x] Plugins > Visualization > Plot (sine wave ok)
- [x] Plugins > Introspection > Node Graph (node graph ok)
- [x] Plugins > Topic > Topic Monitor (topics monitoring ok) 
- [ ] Plugins > Logging > Bag (bag logging not ok)

Verification:
Same `rqt` `rosbag` error described in #1470. Will wait for fix and validate this again.
[image]

closed 2024-06-26 by marcoag