osrf/ros2_test_cases #1471: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Build type: Binary
* Chip set: ARM64
* DDS: FastDDS
* OS: Ubuntu Noble 24.04

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

Tacha-S, 2024-05-03:
Test result

```bash
# Terminal 1
ubuntu@8ab521739af6:/$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-05-03-07-00-28-665799-8ab521739af6-686
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [689]
[INFO] [robot_state_publisher-2]: process started with pid [690]
[INFO] [dummy_joint_states-3]: process started with pid [691]
[INFO] [dummy_laser-4]: process started with pid [692]
[dummy_laser-4] [INFO] [1714719628.729500200] [dummy_laser]: angle inc: 0.004363
[dummy_laser-4] [INFO] [1714719628.729543218] [dummy_laser]: scan size: 1081
[dummy_laser-4] [INFO] [1714719628.729546589] [dummy_laser]: scan time increment:       0.000000
[robot_state_publisher-2] [INFO] [1714719628.732025512] [robot_state_publisher]: Robot initialized
# Terminal 2
ubuntu@8ab521739af6:/$ rqt
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ubuntu'
QLayout::removeWidget: Cannot remove a null widget.
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag_plugins/image_plugin.py", line 35, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rqt_bag_plugins/image_timeline_renderer.py", line 44, in <module>
    from PIL.ImageQt import ImageQt
ImportError: cannot import name 'ImageQt' from 'PIL.ImageQt' (/usr/lib/python3/dist-packages/PIL/ImageQt.py). Did you mean: 'Image'?

rqt_bag.TimelineFrame.load_plugins() failed to load plugin "rqt_bag_plugins/BagImagePlugin":
cannot import name 'ImageQt' from 'PIL.ImageQt' (/usr/lib/python3/dist-packages/PIL/ImageQt.py)
```

Plot, Node Graph and Topic Monitor plugins work fine.
Bag plugin has a problem caused by the version of the pillow package.

Tacha-S, 2024-05-04:
Since support for QT5 has ended, we will need to install `python3-pyqt6`. (ref https://github.com/python-pillow/Pillow/pull/6237)

clalancette, 2024-05-06:
> Since support for QT5 has ended, we will need to install `python3-pyqt6`. (ref [python-pillow/Pillow#6237](https://github.com/python-pillow/Pillow/pull/6237))

It's worth a shot, but I'm not sure that it is going to work with the rest of our stack.  In particular, all of RQT is still written in PyQt5, so I'm not sure it is going to work to just install that package.  We may have to do another workaround.

knmcguire, 2024-05-06:
fair warning though... switching to Qt6 is simpler said than done... we had to also do that last year with our Qt based client and let me tell you... we had to change a lot.

clalancette, 2024-05-06:
> fair warning though... switching to Qt6 is simpler said than done... we had to also do that last year with our Qt based client and let me tell you... we had to change a lot.

Yeah, that's why I we need a workaround for Qt5 for now.  We'll eventually port to Qt6, but that is not going to happen for Jazzy.

closed 2024-06-26 by marcoag