osrf/ros2_test_cases #1472: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* OS: Ubuntu Noble 24.04
* Build type: Binary
* Chip set: ARM64
* DDS: CycloneDDS

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

knmcguire, 2024-05-03:
I'll be looking into this now on a raspberry pi 5, ubuntu nobel, with ROS2 installed as binary with the cyclonedds tag skipped with build

knmcguire, 2024-05-03:
Tried it out with

Terminal 1:
```
$ . ~/ros2_jazzy/ros2-linux/setup.bash
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
[INFO] [launch]: All log files can be found below -----/.ros/log/2024-05-03-14-18-26-979591-bitcraze-RP5-47968
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dummy_map_server-1]: process started with pid [47978]
[INFO] [robot_state_publisher-2]: process started with pid [47979]
[INFO] [dummy_joint_states-3]: process started with pid [47980]
[INFO] [dummy_laser-4]: process started with pid [47981]
[dummy_laser-4] [INFO] [1714738707.198127357] [dummy_laser]: angle inc:	0.004363
[dummy_laser-4] [INFO] [1714738707.198257857] [dummy_laser]: scan size:	1081
[dummy_laser-4] [INFO] [1714738707.198281376] [dummy_laser]: scan time increment: 	0.000000
[robot_state_publisher-2] [INFO] [1714738707.252346086] [robot_state_publisher]: Robot initialized
```

Terminal 2: 
```
bitcraze@bitcraze-RP5:~$ rqt
QSocketNotifier: Can only be used with threads started with QThread
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/qt_gui_cpp/plugin.xml" in package "qt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_py_console/plugin.xml" in package "rqt_py_console" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_shell/plugin.xml" in package "rqt_shell" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/qt_gui_cpp/plugin.xml" in package "qt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_py_console/plugin.xml" in package "rqt_py_console" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_shell/plugin.xml" in package "rqt_shell" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_action/plugin.xml" in package "rqt_action" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_bag/plugin.xml" in package "rqt_bag" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_console/plugin.xml" in package "rqt_console" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_graph/plugin.xml" in package "rqt_graph" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_gui_py/plugin.xml" in package "rqt_gui_py" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_image_view/plugin.xml" in package "rqt_image_view" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_msg/plugin.xml" in package "rqt_msg" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_plot/plugin.xml" in package "rqt_plot" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_publisher/plugin.xml" in package "rqt_publisher" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_reconfigure/plugin.xml" in package "rqt_reconfigure" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_service_caller/plugin.xml" in package "rqt_service_caller" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_srv/plugin.xml" in package "rqt_srv" not found
RosPluginProvider._parse_plugin_xml() plugin file "/opt/ros/jazzy/share/rqt_topic/plugin.xml" in package "rqt_topic" not found
PluginManager._load_plugin() could not load plugin "rqt_bag/Bag": plugin not available
PluginManager._load_plugin() could not load plugin "rqt_graph/RosGraph": plugin not available
PluginManager._load_plugin() could not load plugin "rqt_plot/Plot": plugin not available
PluginManager._load_plugin() could not load plugin "rqt_topic/TopicPlugin": plugin not available
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
```

I was not able to finish the tests as those menu options like plot, node graph, bag and topic monitor weren't available. However, it seems more that there is some conflict of the install as it seems to be looking for the packages in /opt/ros/jazzy/share/rqt_topic/plugin.xml which doesn't exist anymore. Especially since this test does seem to work for fastdds: https://github.com/osrf/ros2_test_cases/issues/1471

knmcguire, 2024-05-06:
It's a wonder what a simple restart can do! It all worked fine now for the exception of the bag functionality, but I've seen that people are already on top of that.

```
$ rqt
QSocketNotifier: Can only be used with threads started with QThread
QLayout::removeWidget: Cannot remove a null widget.
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "...ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "...ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rqt_bag_plugins/image_plugin.py", line 35, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "...ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rqt_bag_plugins/image_timeline_renderer.py", line 44, in <module>
    from PIL.ImageQt import ImageQt
ImportError: cannot import name 'ImageQt' from 'PIL.ImageQt' (/usr/lib/python3/dist-packages/PIL/ImageQt.py). Did you mean: 'Image'?

rqt_bag.TimelineFrame.load_plugins() failed to load plugin "rqt_bag_plugins/BagImagePlugin":
cannot import name 'ImageQt' from 'PIL.ImageQt' (/usr/lib/python3/dist-packages/PIL/ImageQt.py)
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
Wayland does not support QWindow::requestActivate()
This plugin supports grabbing the mouse only for popup windows
/home/bitcraze/ros2_jazzy/ros2-linux/lib/python3.12/site-packages/rqt_plot/data_plot/mat_data_plot.py:120: UserWarning: Tight layout not applied. The left and right margins cannot be made large enough to accommodate all axes decorations.
  self.figure.tight_layout()
This plugin supports grabbing the mouse only for popup windows
```
Screenshot:
[image]

**Seems to work fine!** for the exception of bag plugin but that is not caused by cyclone dds or binary instead of debian.

marcoag, 2024-05-07:
Yup, will close this as the rqt  issue is being worked out ant tracked elsewhere.

Thanks for the test!

closed 2024-05-07 by marcoag