osrf/ros2_test_cases #1474: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Debian
* DDS: FastDDS

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

ganatrask, 2024-05-01:
Hey @marcoag I tested this installation, can you assign this issue to me.

I was able to install following the steps, do I need to provide outputs to get this issue closed? 

Thanks.

marcoag, 2024-05-01:
If you can provide some output that proves the steps were run that would be great, yes!

Thanks for testing!

ganatrask, 2024-05-01:
Sure, I can share some screenshot but it's installation steps, I don't know what should I share. I was able to install and now it there anything that you expect me to share here?

marcoag, 2024-05-01:
Usually just some output/screenshot would work, but I understand that for installation instructions is a bit complicated.

closed 2024-05-01 by marcoag

ganatrask, 2024-05-01:
Got it, 

I think this should work to see. 

[image]

Let me know if you need anything else.