osrf/ros2_test_cases #1475: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* Chip set: AMD64
* DDS: CycloneDDS
* OS: Ubuntu Noble 24.04
* Build type: Debian

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

msadowski, 2024-05-01:
Hope I understood the rules correctly, I'm taking a look at this now, feel free to assign it to me

clalancette, 2024-05-01:
> Hope I understood the rules correctly, I'm taking a look at this now, feel free to assign it to me

Please go ahead.

msadowski, 2024-05-01:
After completing the [Installation](https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debians.html) I was able to run talker and listener (this is before installing Cyclone):

```
mat@mat-ThinkPad-T480:~$ ros2 run demo_nodes_cpp talker
[INFO] [1714594611.464960090] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714594612.465017799] [talker]: Publishing: 'Hello World: 2'
[INFO] [1714594613.464918416] [talker]: Publishing: 'Hello World: 3'
[INFO] [1714594614.464807112] [talker]: Publishing: 'Hello World: 4'
[INFO] [1714594615.465170284] [talker]: Publishing: 'Hello World: 5'
...
```

```
mat@mat-ThinkPad-T480:~$ ros2 run demo_nodes_py listener
[INFO] [1714594619.490983493] [listener]: I heard: [Hello World: 9]
[INFO] [1714594620.466014286] [listener]: I heard: [Hello World: 10]
[INFO] [1714594621.466377915] [listener]: I heard: [Hello World: 11]
...
```

Then, following [this page](http://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html) I was trying to install cyclone using the command:

```
sudo apt install ros-jazzy-rmw-cyclonedds-cpp
```

Setting the $RMW_IMPLEMENTATION to the correct value made it work as epxected:

```
mat@mat-ThinkPad-T480:~$ echo $RMW_IMPLEMENTATION
rmw_cyclonedds_cpp
mat@mat-ThinkPad-T480:~$ ros2 run demo_nodes_cpp talker
[INFO] [1714596134.524656340] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714596135.524408464] [talker]: Publishing: 'Hello World: 2'
[INFO] [1714596136.524416732] [talker]: Publishing: 'Hello World: 3'
[INFO] [1714596137.524825241] [talker]: Publishing: 'Hello World: 4'
```
```
mat@mat-ThinkPad-T480:~$ echo $RMW_IMPLEMENTATION
rmw_cyclonedds_cpp
mat@mat-ThinkPad-T480:~$ ros2 run demo_nodes_cpp listener
[INFO] [1714596134.524972645] [listener]: I heard: [Hello World: 1]
[INFO] [1714596135.524688490] [listener]: I heard: [Hello World: 2]
[INFO] [1714596136.524638385] [listener]: I heard: [Hello World: 3]
[INFO] [1714596137.525006054] [listener]: I heard: [Hello World: 4]
```

The prerequisite page says I should have rosdep installed.  The hyperlink to [rosdep installed](https://wiki.ros.org/rosdep#Installing_rosdep) points to wiki.ros.org, instead, it should probably point to http://docs.ros.org/en/jazzy/Tutorials/Intermediate/Rosdep.html ?

marcoag, 2024-05-02:
Thanks for testing! Not sure how much we want that change but I opened a PR so we can follow up there: https://github.com/ros2/ros2_documentation/pull/4344.

closed 2024-05-02 by marcoag

hwoithe, 2024-05-02:
I apologize if I am posting this in the wrong issue, but this seems to the closest I could find.

I followed the "Build from source code" instructions [here](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html#build-from-source-code) and they did not work for me.

[image]

```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble
```