osrf/ros2_test_cases #1478: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* Chip set: ARM64
* OS: Ubuntu Noble 24.04
* DDS: CycloneDDS
* Build type: Debian

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

knmcguire, 2024-05-03:
I'm currently testing this out on a Raspberry Pi5 with a freshly installed Ubuntu 24.04

knmcguire, 2024-05-03:
I've installed Cyclone DDS by debian packages [via these instructions.](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Working-with-Eclipse-CycloneDDS.html)

Terminal 1:
```
$ ros2 run demo_nodes_cpp talker
$ source /opt/ros/jazzy/setup.bash
$ ros2 run demo_nodes_cpp talker
[INFO] [1714729114.081580579] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714729115.081566138] [talker]: Publishing: 'Hello World: 2'
...
```

Terminal 2:
```
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ source /opt/ros/jazzy/setup.bash
$ ros2 run demo_nodes_cpp listener
[INFO] [1714729135.081895154] [listener]: I heard: [Hello World: 22]
[INFO] [1714729136.081955675] [listener]: I heard: [Hello World: 23]
```

**Seems all okay!** 

Just some comments on the documentation though, which could improve it a bit.

- It were quite some clicks to get to the RMW alternative installs for cycloneDDS. How we replace the [guide link](https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debians.html#install-additional-rmw-implementations-optional) in ['Install additional RMW implementations (optional)'](https://docs.ros.org/en/jazzy/Installation/Ubuntu-Install-Debians.html#install-additional-rmw-implementations-optional) with a direct link to the [installation instructions](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations.html). That already has a link to the earlier mentioned guide.
- For the Cyclone dds instructions, let's perhaps put the full lines for the talker and listener such that it is less copy paste to test out, like in one blob like:
```
export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
source /opt/ros/jazzy/setup.bash
ros2 run demo_nodes_cpp talker
```

closed 2024-05-06 by ahcorde