osrf/ros2_test_cases #1479: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* DDS: ConnextDDS
* Chip set: ARM64
* Build type: Debian
* OS: Ubuntu Noble 24.04

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

slowrunner, 2024-05-02:
Sorry, I am not able to complete this ticket at this time - There is no Ubuntu 24 download available.

clalancette, 2024-05-02:
> Sorry, I am not able to complete this ticket at this time - There is no Ubuntu 24 download available.

Can you clarify what you mean here?  There should be Jazzy debian packages available for Ubuntu 24.04 aarch64.

slowrunner, 2024-05-02:
This ticket requires connextdds be installed - I cannot find an aarch64 connextdds installation package for Ubuntu 24.04:

```
dpkg -l | grep connext
ii  rti-connext-dds-6.0.1                         0.0.0-0                                  arm64        Trivial package standing in for rti-connext-dds on arm64

```

The package is empty.  The page https://docs.ros.org/en/jazzy/Installation/DDS-Implementations.html states:

"Official binary packages from RTI[](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations.html#official-binary-packages-from-rti)
You can install the Connext 6.0.1 package for Linux provided by RTI, via options available for [university, purchase or evaluation](https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-University-Eval.html)"

Which contains a link to https://docs.ros.org/en/jazzy/Installation/DDS-Implementations/Install-Connext-University-Eval.html

Which took me to a signup page: https://www.rti.com/free-trial

Which allowed me to a download page: https://www.rti.com/free-trial/dds-files-tele

Which only has Ubuntu 22 versions or aarch64 - no Noble package available.

clalancette, 2024-05-02:
Ah yes, thank you for clarifying.  In this case, we don't actually have Connext debians available for aarch64 either, so we actually just build a "dummy" package here.  So this ticket can be closed since this is something we can't test.

Thank you for looking at it anyway!

closed 2024-05-02 by clalancette