osrf/ros2_test_cases #1480: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* Build type: Binary
* DDS: FastDDS
* Chip set: AMD64
* OS: Ubuntu Noble 24.04

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

Kartik9250, 2024-05-03:
I would like to work on this

Kartik9250, 2024-05-03:
I tried the whole installation procedure, everything works and seems good. One thing I would like to add is that It would help beginners if there was line suggesting that they can add `source /opt/ros/jazzy/setup.bash` in their user shell config.

closed 2024-05-03 by clalancette

Kartik9250, 2024-05-03:
@clalancette I forgot to add two things:
- ```[rti_connext_dds_cmake_module][warning] No RTI Connext DDS installation specified.. RTI Connext DDS will not be available at runtime,unless you already configured LD_LIBRARY_PATH manually.``` This shows up everytime we source the setup.zsh. Its just a warning for connext_dds I suppose, I was using the default fastdds.
- In the Jazzy's Testing with pre release binary section, the download link at line no.76 takes us to the binary releases of rolling. I installed the binary by the tar provided at the ros2 github releases page. So that link should be mentioned there.

mikaelarguedas, 2024-05-06:
> One thing I would like to add is that It would help beginners if there was line suggesting that they can add source /opt/ros/jazzy/setup.bash in their user shell config.

When we install from the fat binary the archive is not extracted to /opt/ros/jazzy is it ? :thinking:

Kartik9250, 2024-05-06:
> > One thing I would like to add is that It would help beginners if there was line suggesting that they can add source /opt/ros/jazzy/setup.bash in their user shell config.
> 
> When we install from the fat binary the archive is not extracted to /opt/ros/jazzy is it ? 🤔

Yeah it isn't my mistake, the path is different. Basically I was just giving suggestion.

clalancette, 2024-05-06:
> * In the Jazzy's Testing with pre release binary section, the download link at line no.76 takes us to the binary releases of rolling. I installed the binary by the tar provided at the ros2 github releases page. So that link should be mentioned there.

We've fixed this now, so it should point to the jazzy binaries.