osrf/ros2_test_cases #1482: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Binary
* DDS: ConnextDDS

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

mikaelarguedas, 2024-05-06:
Tested successfully within ubuntu:noble container :+1:

Unclear parts:

Not clear for user that you need to install `rti-connext-dds-6.0.1` to get it to run

Maybe personal but the rmw implementation library names are confusing:
For Fast DDS -> rmw_fastrtps_cpp  (and not rmw_fastdds_cpp)
For Cyclone DDS -> rmw_cyclonedds_cpp
For Connext DDS -> rmw_connext (without the `_cpp`)

clalancette, 2024-05-06:
> Not clear for user that you need to install `rti-connext-dds-6.0.1` to get it to run

Oh, right.  For the binary install, you need to install that separately (it comes in automatically when you use Debian packages).  I'd definitely be for adding something explaining how to do that, but I don't really know where we would add it.  Do you have any suggestions?

mikaelarguedas, 2024-05-06:
I made an attempt here https://github.com/ros2/ros2_documentation/pull/4369/commits/a6bc319087487470a4e3a32187e20ea5a1380471 WDYT ?

clalancette, 2024-05-07:
Since we are iterating on the fix over in the ros2_documentation PR, and this test was otherwise successful, I'm going to close this out.  Thanks for testing.

closed 2024-05-07 by clalancette