osrf/ros2_test_cases #1483: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* Build type: Binary
* Chip set: ARM64
* DDS: FastDDS
* OS: Ubuntu Noble 24.04

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

clalancette, 2024-05-02:
> My only comment is the strangeness of the references to Rolling on the Jazzy Install pages:

Yep, you are totally right about that.  That's because we fork the Jazzy documentation from Rolling, and we forgot to update that.  I'll open a PR with those fixes here shortly.

slowrunner, 2024-05-02:
Can I be assigned this ticket - I'll complete the uninstall section and report completion.

slowrunner, 2024-05-02:
Oops, this is "binary" I did "Debian" - I was rushing, coffee will wear off soon.

clalancette, 2024-05-02:
All right, shortly == 4 hours, but I've now opened the PR that should fix the wording in the documentation: https://github.com/ros2/ros2_documentation/pull/4353

slowrunner, 2024-05-02:
This install resulted in a successful ROS 2 Jazzy environment - although I got temporarily got confused after finding the image to download.

https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Install-Binary.html:
"Binary releases of Rolling Ridley are not provided. Instead you may download nightly [prerelease binaries](https://docs.ros.org/en/jazzy/Installation/Testing.html#prerelease-binaries)."

That page had me go to https://ci.ros2.org/view/packaging/  
which had me click on "Packaging-Linux-aarch64" which took me to 
https://ci.ros2.org/view/packaging/job/packaging_linux-aarch64/

which I downloaded with wget, then instead of returning to the original install instructions, I continued with the instructions on https://docs.ros.org/en/jazzy/Installation/Testing.html#prerelease-binaries  which just had me source the setup.bash.

Once I returned to the original install binaries page https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Install-Binary.html and followed the instructions there, I ended up with a successful install.

I assume this page will point to somewhere different once released binaries are built and published.

Can close this ticket, for the time being.  I think the install instructions should be retested after the release binaries are available.

clalancette, 2024-05-06:
We ended up improving the installation instructions, and this was otherwise successful, so closing this out.

Thanks for testing!

closed 2024-05-06 by clalancette