osrf/ros2_test_cases #1484: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* OS: Ubuntu Noble 24.04
* Build type: Binary
* Chip set: ARM64
* DDS: CycloneDDS

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

knmcguire, 2024-05-03:
Currently working on this issue with a Rapsberry 5 with Ubuntu Noble. It's pretty fresh but I just have installed and deinstalled ROS2 through debian so I hope that the removal was as clean as it could.

clalancette, 2024-05-15:
> I did install the cyclone dds through debian packages, which also installed a whole bunch of other packages that it depends on... so not sure if that is what I was supposed to do.

Yep, that's correct.

> When I removed the cyclonedds from the skip keys, the standard tests like I did here above worked as well but it seemed to have grabbed the bindings for rolling and not jazzy

Oh, hm.  So you are saying it installed `ros-rolling-cyclonedds` via rosdep?  Did it install anything else?

knmcguire, 2024-05-16:
ah no sorry I don't remember which other packages it installed. Let me try again tomorrow on the RP 5 and I can confirm.

But other than that, I just went out of the way to try this. If I just follow the instructions on the word it all works well so you can close this issue for that

clalancette, 2024-05-16:
> But other than that, I just went out of the way to try this. If I just follow the instructions on the word it all works well so you can close this issue for that

Fantastic, thanks.  Will close for now, feel free to reopen or comment again if you find something wonky.

closed 2024-05-16 by clalancette