osrf/ros2_test_cases #1490: Local workspace overlay

marcoag, 2024-04-30:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Configuration

* Chip set: ARM64
* OS: Ubuntu Noble 24.04
* DDS: CycloneDDS
* Build type: Debian

## Process

    
            
### Local workspace overlay works

pradyum, 2024-05-03:
## Test Environment

```
   PLATFORM INFORMATION
system           : Linux
platform info    : Linux-6.8.0-31-generic-aarch64-with-glibc2.39
release          : 6.8.0-31-generic
processor        : aarch64

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

I am able to build a package which is depended on rcpcpp and sensor_msgs. The build completed successfully and able to source this local workspace to run the executable.

marcoag, 2024-05-03:
Thanks for testing!

closed 2024-05-03 by marcoag