osrf/ros2_test_cases #1493: Local workspace overlay

marcoag, 2024-04-30:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Configuration

* OS: Ubuntu Noble 24.04
* DDS: CycloneDDS
* Build type: Binary
* Chip set: AMD64

## Process

    
            
### Local workspace overlay works

mikaelarguedas, 2024-05-04:
I could run https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html without issues

:heavy_check_mark: test passed

Question:
Shouldnt this be RMW agnostic and thus gernerate less test case ? looks identical to https://github.com/osrf/ros2_test_cases/issues/1486 to me

mikaelarguedas, 2024-05-06:
Tested successfully with minors fixup available at https://github.com/ros2/ros2_documentation/pull/4369

closed 2024-05-06 by clalancette