osrf/ros2_test_cases #1494: Local workspace overlay

marcoag, 2024-04-30:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Configuration

* OS: Ubuntu Noble 24.04
* Chip set: AMD64
* Build type: Binary
* DDS: ConnextDDS

## Process

    
            
### Local workspace overlay works

mikaelarguedas, 2024-05-06:
Tested successfully with minor fixup at https://github.com/ros2/ros2_documentation/pull/4369

clalancette, 2024-05-06:
Since we have the open PR, going to close this as otherwise successful.  Thanks for testing!

closed 2024-05-06 by clalancette