osrf/ros2_test_cases #1496: Local workspace overlay

marcoag, 2024-04-30:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Configuration

* OS: Ubuntu Noble 24.04
* Build type: Binary
* Chip set: ARM64
* DDS: CycloneDDS

## Process

    
            
### Local workspace overlay works

knmcguire, 2024-05-03:
Currently working on this with a raspberry pi 5, Ubuntu Noble and binary install of ROS with cyclonedds removed from the skip-tags

clalancette, 2024-05-03:
> WARNING: given --rosdistro jazzy but ROS_DISTRO is "rolling". Ignoring environment.

Ah, yeah.  This is actually a mistake in the documentation.  https://docs.ros.org/en/jazzy/Installation/Alternatives/Ubuntu-Install-Binary.html#install-ros-2 should actually say to download from the pre-release section of https://github.com/ros2/ros2/releases .  I'll update that.

clalancette, 2024-05-03:
All right, here is the PR: https://github.com/ros2/ros2_documentation/pull/4356 .  Once that is reviewed, I'll merge it.  I'm going to close this out, thanks for testing!

closed 2024-05-03 by clalancette