osrf/ros2_test_cases #1504: ros2cli network tests

marcoag, 2024-04-30:
Test the basic functionality of the ros2cli package family in an environment where nodes are running on other machines on the network.

## Configuration

* DDS: FastDDS
* OS: Windows
* Build type: Binary
* Chip set: AMD64

## Links

* [Repo URL](https://github.com/ros2/ros2cli)

## Process

    
            
### Step 1 - Node discovery
            
        

    
With another node running, use `ros2 node list` to ensure that the expected node is listed. Then stop the node and ensure that the list operation no longer shows the node.

        

    

    
    
            
### Step 2 - Topic discovery
            
        

    
With another node running, use `ros2 topic list` to ensure that the expected topics are listed. Then stop the node and ensure that the list operation no longer shows the node's topics.

        

    

    
    
            
### Step 3 - Service discovery
            
        

    
With another node running, use `ros2 service list` to ensure that the expected services are listed. Then stop the node and ensure that the list operation no longer shows the node's services.

        

    

    
    
            
### Step 4 - Parameter discovery
            
        

    
With another node running, use `ros2 param list` to ensure that the expected parameters are listed. Then stop the node and ensure that the list operation no longer shows the node's parameters.

Trushant-Adeshara-UM, 2024-05-14:
I would like to take up this ticket, can you please assign it to me?

Trushant-Adeshara-UM, 2024-05-14:
### Environment:

OS: Windows 11
Steps to Reproduce and Observations:

### Step 1 - Node discovery
It can be seen in the following snapshot that ```/talker``` node is running using command ```ros2 node list```.

[image]

### Step 2 - Topic discovery
It can be seen in the following snapshot that ```/talker``` node uses topic ```/chatter``` to publish data.

[image]

### Step 3 - Service discovery
It can be seen in the following snapshot that ```/talker``` node has following set of services associated with it which can be seen while it is running:
- /talker/describe_parameters
- /talker/get_parameter_types
- /talker/get_parameters
- /talker/get_type_description
- /talker/list_parameters
- /talker/set_parameters
- /talker/set_parameters_atomically

[image]

### Step 4 - Parameter discovery
It can be seen in the following snapshot that ```/talker``` node has following set of parameters associated with it which can be seen while it is running:
- /talker:
	qos_overrides./parameter_events.publisher.depth
	qos_overrides./parameter_events.publisher.durability
	qos_overrides./parameter_events.publisher.history
	qos_overrides./parameter_events.publisher.reliability
	start_type_description_service
	use_sim_time

[image]

marcoag, 2024-05-14:
Thanks for testing!

closed 2024-05-14 by marcoag