osrf/ros2_test_cases #1516: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* DDS: FastDDS
* OS: Windows
* Build type: Binary
* Chip set: AMD64

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

#### Errors

There are errors with bag plugin. This saved bag file has 0.0s duration with same start and end timestamps.

```
PS C:\Users\pc\Desktop> ros2 bag info .\2024-05-05-06-01-15\

Files:             2024-05-05-06-01-15_0.mcap
Bag size:          1.8 KiB
Storage id:        mcap
ROS Distro:        jazzy
Duration:          0.0s
Start:             Apr 11 2262 16:47:16.854 (9223372036.854)
End:               Apr 11 2262 16:47:16.854 (9223372036.854)
Messages:          0
Topic information:
Service:           0
Service information:
```

Other checks are passed except bag plugin.

marcoag, 2024-06-26:
Closing as it's been more than a month since the end of the tutorial party. We'll keep track of the errors reported here and see what can be done. Thanks for the work @pradyum!

closed 2024-06-26 by marcoag