osrf/ros2_test_cases #1517: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* OS: Windows
* Chip set: AMD64
* DDS: CycloneDDS
* Build type: Binary

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

pradyum, 2024-05-05:
## Setup

#### Installation Changes

1. openssl v1.1.1 - downloaded through choco

```
   PLATFORM INFORMATION
system           : Windows
platform info    : Windows-10-10.0.19041-SP0
release          : 10
processor        : Intel64 Family 6 Model 154 Stepping 4, GenuineIntel

   QOS COMPATIBILITY LIST
compatibility status    : No publisher/subscriber pairs found

   RMW MIDDLEWARE
middleware name    : rmw_cyclonedds_cpp

   ROS 2 INFORMATION
distribution name      : jazzy
distribution type      : ros2
distribution status    : pre-release
release platforms      : {'debian': ['bookworm'], 'rhel': ['9'], 'ubuntu': ['noble']}
```
## Result

#### Error
bag plugin is not working properly which is addressed in previous rqt issues.

```
C:\>rqt
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_jazzy\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
RosPluginProvider._parse_plugin_xml() plugin file "C:\dev\ros2_jazzy\share\rqt_gui_cpp/plugin.xml" in package "rqt_gui_cpp" not found
Mesa: SVGA3D; build: RELEASE;

Mesa: mesa-21.3.8
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_gui\ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_bag_plugins\image_plugin.py", line 35, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_bag_plugins\image_timeline_renderer.py", line 44, in <module>
    from PIL.ImageQt import ImageQt
ImportError: cannot import name 'ImageQt' from 'PIL.ImageQt' (C:\Python38\lib\site-packages\PIL\ImageQt.py)

rqt_bag.TimelineFrame.load_plugins() failed to load plugin "rqt_bag_plugins/BagImagePlugin":
cannot import name 'ImageQt' from 'PIL.ImageQt' (C:\Python38\lib\site-packages\PIL\ImageQt.py)
[INFO] [1714922925.908536100] [rqt_bag.BagWidget]: Recording to C:/Users/pc/Desktop/2024-05-05-08-28-41.
Error opening bag for recording [C:/Users/pc/Desktop/2024-05-05-08-28-41]: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. rosbag2_py._storage.TopicMetadata(id: int, name: str, type: str, serialization_format: str, offered_qos_profiles: List[rosbag2_py._storage.QoS] = [], type_description_hash: str = '')

Invoked with: kwargs: name='/joint_states', type='sensor_msgs/msg/JointState', serialization_format='cdr', offered_qos_profiles='- history: 1\n  depth: 10\n  reliability: 1\n  durability: 2\n  lifespan:\n    sec: 9223372036\n    nsec: 854775807\n  deadline:\n    sec: 9223372036\n    nsec: 854775807\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 9223372036\n    nsec: 854775807\n  avoid_ros_namespace_conventions: false\n'
[WARN] [1714922943.124774300] [rosbag2_storage_mcap]: attempted to read in receive timestamp order with no message index
[WARN] [1714922943.126459900] [rosbag2_cpp]: No topics were listed in metadata.
Traceback (most recent call last):
  File "C:\dev\ros2_jazzy\Lib\site-packages\rclpy\time.py", line 87, in __add__
    return Time(
  File "C:\dev\ros2_jazzy\Lib\site-packages\rclpy\time.py", line 57, in __init__
    raise OverflowError(
OverflowError: Total nanoseconds value is too large to store in C time point.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_bag\bag_widget.py", line 281, in _handle_load_clicked
    self.load_bag(filename)
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_bag\bag_widget.py", line 318, in load_bag
    self._timeline.add_bag(bag)
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_bag\bag_timeline.py", line 175, in add_bag
    self._timeline_frame.reset_timeline()
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_bag\timeline_frame.py", line 1042, in reset_timeline
    self.reset_zoom()
  File "C:\dev\ros2_jazzy\Lib\site-packages\rqt_bag\timeline_frame.py", line 1075, in reset_zoom
    end_stamp = start_stamp + Duration(seconds=5)
  File "C:\dev\ros2_jazzy\Lib\site-packages\rclpy\time.py", line 91, in __add__
    raise OverflowError('Addition leads to overflow in C storage.') from e
OverflowError: Addition leads to overflow in C storage.
```

Except bag plugin other checks are working as expected.

marcoag, 2024-06-26:
Closing as it's been more than a month since the tutorial party ended. Keeping track of this so we can see possible improvements.