osrf/ros2_test_cases #1522: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* DDS: FastDDS
* OS: Windows
* Build type: Binary
* Chip set: AMD64

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

jaspreet491, 2024-05-01:
Testing the Windows binary install. Please assign this to me.

jaspreet491, 2024-05-01:
Issue - Broken install link.
The link to the install page for installing OpenSSL version Win64 OpenSSL v1.1.1n is broken/installer unavailable.
Only installer/binaries for v3.x.x are available on the linked page.

Installation instruction step -  [Install OpenSSL](https://docs.ros.org/en/jazzy/Installation/Windows-Install-Binary.html#install-openssl) 

Probably already known but here's a note from [OpenSSL website](https://www.openssl.org/source/) 
Note: The latest stable version is the 3.2 series supported until 23rd November 2025. Also available is the 3.1 series supported until 14th March 2025, and the 3.0 series which is a Long Term Support (LTS) version and is supported until 7th September 2026. All older versions (including 1.1.1, 1.1.0, 1.0.2, 1.0.0 and 0.9.8) are now out of support and should not be used. Users of these older versions are encouraged to upgrade to 3.2 or 3.0 as soon as possible.

clalancette, 2024-05-01:
Thanks.  We'll have to figure out how to make things work with OpenSSL 3.X series.  We're also tracking this in https://github.com/osrf/ros2_test_cases/issues/1526

jaspreet491, 2024-05-01:
Issue - Incorrect/Ambiguous Path while installing xmllint ([Install Miscellaneous Prerequisites](https://docs.ros.org/en/jazzy/Installation/Windows-Install-Binary.html#install-miscellaneous-prerequisites)) - 

The following step asks to add `C:\xmllint\bin` to the path
> Add C:\xmllint\bin to the PATH

But the above path doesn't exist.
Here is what the path `C:\xmllint\` looks like after extracting 64bit binaries of [libxml2-2.9.3-win32-x86_64.7z](https://www.zlatkovic.com/pub/libxml/64bit/libxml2-2.9.3-win32-x86_64.7z), [iconv-1.14-win32-x86_64.7z](https://www.zlatkovic.com/pub/libxml/64bit/iconv-1.14-win32-x86_64.7z) and [zlib-1.2.8-win32-x86_64.7z](https://www.zlatkovic.com/pub/libxml/64bit/zlib-1.2.8-win32-x86_64.7z) into the `C:\xmllint` as mentioned in
the previous step - 
[image]

Each folder contains one or more subfolders like `bin`, `include`, `lib` and `share`.

prototriangle, 2024-05-01:
I think this is more due to ambiguity in the instructions.
> Unpack all archives into e.g. C:\xmllint

The goal (I think) is to extract the contents each to the same space to get something like this:
[image]

jaspreet491, 2024-05-02:
Merged content of [iconv-1.14-win32-x86_64.7z](https://www.zlatkovic.com/pub/libxml/64bit/iconv-1.14-win32-x86_64.7z) and [zlib-1.2.8-win32-x86_64.7z](https://www.zlatkovic.com/pub/libxml/64bit/zlib-1.2.8-win32-x86_64.7z) into [libxml2-2.9.3-win32-x86_64.7z](https://www.zlatkovic.com/pub/libxml/64bit/libxml2-2.9.3-win32-x86_64.7z) under path `C:\xmllint`.
[image]

jaspreet491, 2024-05-02:
Issue - Call to setup workspace under [Setup Environment](https://docs.ros.org/en/jazzy/Installation/Windows-Install-Binary.html#setup-environment) step using `call C:\dev\ros2_jazzy\local_setup.bat` prints a warning 
> C:\Windows\system32>call C:\ros2_jazzy_dev\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."

Also,
> It is normal that the previous command, if nothing else went wrong, outputs “The system cannot find the path specified.” exactly once.

is confusing because unintentionally typing a non-existent path also outputs `The system cannot find the path specified. ` just once. 

> C:\Windows\system32>call D:\ros2_jazzy_dev\ros2-windows\local_setup.bat
The system cannot find the path specified.

Furthermore, talker and listener demo gives the following error - 

> Microsoft Windows [Version 10.0.19045.4291]
(c) Microsoft Corporation. All rights reserved.
C:\Windows\system32>call C:\ros2_jazzy_dev\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."
C:\Windows\system32>ros2 run demo_nodes_cpp talker
[ros2run]: Process exited with failure 3221226505
C:\Windows\system32>

> Microsoft Windows [Version 10.0.19045.4291]
(c) Microsoft Corporation. All rights reserved.
C:\Windows\system32>call C:\ros2_jazzy_dev\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."
C:\Windows\system32>ros2 run demo_nodes_py listener
Traceback (most recent call last):
  File "C:\ros2_jazzy_dev\ros2-windows\lib\demo_nodes_py\listener-script.py", line 33, in <module>
    sys.exit(load_entry_point('demo-nodes-py==0.33.2', 'console_scripts', 'listener')())
  File "C:\ros2_jazzy_dev\ros2-windows\Lib\site-packages\demo_nodes_py\topics\listener.py", line 33, in main
    rclpy.init(args=args)
  File "C:\ros2_jazzy_dev\ros2-windows\Lib\site-packages\rclpy\__init__.py", line 89, in init
    return context.init(args, domain_id=domain_id)
  File "C:\ros2_jazzy_dev\ros2-windows\Lib\site-packages\rclpy\context.py", line 99, in init
    self.__context = _rclpy.Context(
rclpy._rclpy_pybind11.RCLError: Failed to initialize init options: failed to load any RMW implementations, at C:\ci\ws\src\ros2\rmw_implementation\rmw_implementation\src\functions.cpp:125, at C:\ci\ws\src\ros2\rcl\rcl\src\rcl\init_options.c:75
[ros2run]: Process exited with failure 1
C:\Windows\system32>

jaspreet491, 2024-05-02:
Environment variables including Path using `set` command

> C:\Windows\system32>set
ALLUSERSPROFILE=C:\ProgramData
AMENT_PREFIX_PATH=C:\ros2_jazzy_dev\ros2-windows
APPDATA=C:\Users\Admin\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=133590629475915198
CMAKE_PREFIX_PATH=C:\ros2_jazzy_dev\ros2-windows\opt/yaml_cpp_vendor;C:\ros2_jazzy_dev\ros2-windows\opt/pybind11_vendor;C:\ros2_jazzy_dev\ros2-windows\opt/orocos_kdl_vendor;C:\ros2_jazzy_dev\ros2-windows\opt/gz_math_vendor;C:\ros2_jazzy_dev\ros2-windows\opt/gz_utils_vendor;C:\ros2_jazzy_dev\ros2-windows\opt/gz_cmake_vendor;C:\ros2_jazzy_dev\ros2-windows\opt/console_bridge_vendor;C:\ros2_jazzy_dev\ros2-windows
COLCON_PREFIX_PATH=C:\ros2_jazzy_dev\ros2-windows\
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
ComSpec=C:\Windows\system32\cmd.exe
DriverData=C:\Windows\System32\Drivers\DriverData
OpenCV_DIR=C:\opencv
OPENSSL_CONF=C:\Program Files\OpenSSL-Win64\bin\openssl.cfg
OS=Windows_NT
Path=C:\ros2_jazzy_dev\ros2-windows\opt/zstd_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/zstd_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/yaml_cpp_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/yaml_cpp_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/uncrustify_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/sqlite3_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/sqlite3_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/spdlog_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/spdlog_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/rviz_ogre_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/rviz_ogre_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/rviz_assimp_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/rviz_assimp_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/orocos_kdl_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/mimick_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/libyaml_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/libyaml_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/liblz4_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/liblz4_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/libcurl_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/libcurl_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/gz_math_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/gz_math_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/gz_utils_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/gz_utils_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/gz_cmake_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/gz_cmake_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\opt/console_bridge_vendor/lib;C:\ros2_jazzy_dev\ros2-windows\opt/console_bridge_vendor/bin;C:\ros2_jazzy_dev\ros2-windows\Scripts;C:\ros2_jazzy_dev\ros2-windows\bin;C:\Python38\Scripts\;C:\Python38\;C:\Program Files (x86)\VMware\VMware Player\bin\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\ProgramData\chocolatey\bin;C:\Program Files\OpenSSL-Win64\bin\;C:\opencv\x64\vc16\bin;C:\Program Files\CMake\bin;C:\ProgramData\chocolatey\lib\cunit\lib;C:\ProgramData\chocolatey\lib\tinyxml2\lib;C:\xmllint\bin;C:\Program Files\Graphviz\bin;C:\Users\Admin\AppData\Local\Microsoft\WindowsApps
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC;.PY;.PYW
PKG_CONFIG_PATH=C:\ros2_jazzy_dev\ros2-windows\lib\pkgconfig
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 61 Stepping 4, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=3d04
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
PYTHONPATH=C:\ci\ws\install\lib\python3.8\dist-packages;C:\ros2_jazzy_dev\ros2-windows\Lib\site-packages
Qt5_DIR=C:\Qt\Qt5.12.12\5.12.12\msvc2017_64
QT_QPA_PLATFORM_PLUGIN_PATH=C:\Qt\Qt5.12.12\5.12.12\msvc2017_64\plugins\platforms
ROS_AUTOMATIC_DISCOVERY_RANGE=SUBNET
ROS_DISTRO=jazzy
ROS_PYTHON_VERSION=3
ROS_VERSION=2

Not sure why some of the path contains a mix of forward and backslashes though e.g. everywhere with `\opt/`.

jaspreet491, 2024-05-03:
Got it to work with OpenSSL v1.1.1w.

> C:\Windows\system32>call C:\ros2_jazzy_dev\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."
C:\Windows\system32>ros2 run demo_nodes_cpp talker
[INFO] [1714763331.997426800] [talker]: Publishing: 'Hello World: 1'
[INFO] [1714763333.003903700] [talker]: Publishing: 'Hello World: 2'
[INFO] [1714763334.003887300] [talker]: Publishing: 'Hello World: 3'
[INFO] [1714763335.003965100] [talker]: Publishing: 'Hello World: 4'
[INFO] [1714763336.003821800] [talker]: Publishing: 'Hello World: 5'
[INFO] [1714763337.003558000] [talker]: Publishing: 'Hello World: 6'
[INFO] [1714763338.003697900] [talker]: Publishing: 'Hello World: 7'
[INFO] [1714763339.003815700] [talker]: Publishing: 'Hello World: 8'
[INFO] [1714763340.004072500] [talker]: Publishing: 'Hello World: 9'
[INFO] [1714763341.004061500] [talker]: Publishing: 'Hello World: 10'
[INFO] [1714763342.004103400] [talker]: Publishing: 'Hello World: 11'
[INFO] [1714763343.004350200] [talker]: Publishing: 'Hello World: 12'
[INFO] [1714763344.004385700] [talker]: Publishing: 'Hello World: 13'
[INFO] [1714763345.004576300] [talker]: Publishing: 'Hello World: 14'
[INFO] [1714763346.004630200] [talker]: Publishing: 'Hello World: 15'
[INFO] [1714763347.004654800] [talker]: Publishing: 'Hello World: 16'
[INFO] [1714763348.004722900] [talker]: Publishing: 'Hello World: 17'
[INFO] [1714763349.004574100] [talker]: Publishing: 'Hello World: 18'
[INFO] [1714763350.004933600] [talker]: Publishing: 'Hello World: 19'
[INFO] [1714763351.004842600] [talker]: Publishing: 'Hello World: 20'

> C:\Windows\system32>call C:\ros2_jazzy_dev\ros2-windows\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."
C:\Windows\system32>ros2 run demo_nodes_py listener
[INFO] [1714763339.157000200] [listener]: I heard: [Hello World: 8]
[INFO] [1714763340.027102300] [listener]: I heard: [Hello World: 9]
[INFO] [1714763341.012827600] [listener]: I heard: [Hello World: 10]
[INFO] [1714763342.012803000] [listener]: I heard: [Hello World: 11]
[INFO] [1714763343.014073100] [listener]: I heard: [Hello World: 12]
[INFO] [1714763344.013243200] [listener]: I heard: [Hello World: 13]
[INFO] [1714763345.013258100] [listener]: I heard: [Hello World: 14]
[INFO] [1714763346.013269300] [listener]: I heard: [Hello World: 15]
[INFO] [1714763347.013491400] [listener]: I heard: [Hello World: 16]
[INFO] [1714763348.013468400] [listener]: I heard: [Hello World: 17]
[INFO] [1714763349.013160200] [listener]: I heard: [Hello World: 18]
[INFO] [1714763350.013560000] [listener]: I heard: [Hello World: 19]
[INFO] [1714763351.013607600] [listener]: I heard: [Hello World: 20]
[INFO] [1714763352.014238300] [listener]: I heard: [Hello World: 21]
[INFO] [1714763353.027971300] [listener]: I heard: [Hello World: 22]
[INFO] [1714763354.015590500] [listener]: I heard: [Hello World: 23]
[INFO] [1714763355.016493100] [listener]: I heard: [Hello World: 24]
[INFO] [1714763356.017346700] [listener]: I heard: [Hello World: 25]
[INFO] [1714763357.017336500] [listener]: I heard: [Hello World: 26]
[INFO] [1714763358.016120200] [listener]: I heard: [Hello World: 27]
[INFO] [1714763359.015473800] [listener]: I heard: [Hello World: 28]

Also needed to allow Python and talker through Windows firewall.

jaspreet491, 2024-05-03:
Final Comments - 

- In Step Install Visual Studio probably the picture can be updated to reflect both _Desktop development with C++_ **selected** and  _C++ Cmake tools for Windows_ **unselected**.
- Add OpenSSL to Path 
> In the resulting dialog, click “Environment Variables”, then click “Path” on the bottom pane, finally click “Edit” and add the path below.

Instead of bottom pane specify `Path` under System Variables & not the one under User variables.
And instead of EDIT use NEW to add a new Path in a free space, using edit causes editing of already existing path.

- Installing Miscellaneous pre-requisites - Extract libxml2  (and its dependencies iconv and zlib) into C:\xmllint and merge `bin`, `include`, `lib` and `share` folders. Use 7zip to extract the files.
- Probably remove `The system cannot find the path specified`  warning as it is misleading, refer one of the above comments.
- Ticket for OpenSSL `v3.x.x`. instead of` v1.1.1`, as it is not working with the latest OpenSSL release. Or provide OpenSSL ` v1.1.1` binary directly.

@marcoag is this done or should I also do some tutorials and test debug build?

marcoag, 2024-06-26:
@jaspreet491 thanks for the work here! We'll keep track of the info here, also feel free to open a PR with any docs improvements you might have found. Closing this one for now as we are over a month after the tutorial party was done.

closed 2024-06-26 by marcoag