osrf/ros2_test_cases #1526: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* Build type: Sounce
* DDS: CycloneDDS
* Chip set: AMD64
* OS: Windows

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

prototriangle, 2024-05-01:
I'd like to give this a go on Windows 10 22H2

prototriangle, 2024-05-01:
The [Install OpenSSL](https://docs.ros.org/en/jazzy/Installation/Alternatives/Windows-Development-Setup.html#install-openssl) section has an issue in the instructions:
> Download the Win64 OpenSSL v1.1.1n OpenSSL installer from [this page](https://slproweb.com/products/Win32OpenSSL.html). Scroll to the bottom of the page and download Win64 OpenSSL v1.1.1t. Don’t download the Win32 or Light versions, or the v3.X.Y installers.

[Shining Light Productions](https://slproweb.com/products/Win32OpenSSL.html) currently only offers versions `3.0.13`, `3.1.5`, `3.2.1`, and `3.3.0`.

Would using `v3.X.Y` lead to problems?

clalancette, 2024-05-01:
> Would using `v3.X.Y` lead to problems?

Ah, that is a good question.  I didn't realize they stopped offering the 1.X series.

I think the 3.X series *should* work.  I'm going to suggest you go ahead and try out the latest (3.3.0), and see if it builds and works.  If so, we can update the documentation.

prototriangle, 2024-05-01:
Putting this here for reference: https://github.com/ros2/ros2_documentation/pull/2878

prototriangle, 2024-05-01:
Separate issue has occurred.

Error building `gz_cmake_vendor`

Perhaps an issue with CMake configuration

```
C:\jazzy>colcon build --merge-install
...
Summary: 83 packages finished [8min 33s]
  1 package failed: gz_cmake_vendor
  7 packages aborted: cyclonedds foonathan_memory_vendor libcurl_vendor mcap_vendor rviz_assimp_vendor rviz_ogre_vendor uncrustify_vendor
  272 packages not processed
WNDPROC return value cannot be converted to LRESULT
TypeError: WPARAM is simple, so must be an int object (got NoneType)
```
```
C:\jazzy>colcon build --merge-install --packages-select gz_cmake_vendor --event-handlers console_direct+
Starting >>> gz_cmake_vendor
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Performing install step for 'gz_cmake_vendor'
  Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
  Copyright (C) Microsoft Corporation. All rights reserved.

    1>
    -- Install configuration: "Release"
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/msvcp140.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/msvcp140_1.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/msvcp140_2.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/msvcp140_atomic_wait.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/msvcp140_codecvt_ids.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/vcruntime140_1.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/vcruntime140.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin/concrt140.dll
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/bin
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/benchmark
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/benchmark/run_benchmarks.py
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/codecheck
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/codecheck/cpplint.py
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/codecheck/header_guard.rule
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/codecheck/namespace_AZ.rule
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/tools
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/tools/check_test_ran.py
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/gz/gz-cmake3/tools/doc_check.sh
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/cmake/gz-cmake3/gz-cmake3-config.cmake
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/cmake/gz-cmake3/gz-cmake3-config-version.cmake
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/lib/pkgconfig/gz-cmake3.pc
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/cmake/gz-cmake3/gz-cmake3-utilities-targets.cmake
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/gz
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/gz/utilities
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/gz/utilities/detail
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/gz/utilities/detail/ExtraTestMacros.hh
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/gz/utilities/detail/SuppressWarning.hh
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/gz/utilities/ExtraTestMacros.hh
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/gz/utilities/SuppressWarning.hh
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/ignition
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/ignition/utilities
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/ignition/utilities/ExtraTestMacros.hh
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/include/gz/cmake3/ignition/utilities/SuppressWarning.hh
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/cmake/gz-cmake3/cmake3/FindAVCODEC.cmake
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/cmake/gz-cmake3/cmake3/FindAVDEVICE.cmake
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/cmake/gz-cmake3/cmake3/FindAVFORMAT.cmake
    -- Up-to-date: C:/jazzy/build/gz_cmake_vendor/gz_cmake_vendor-prefix/install/share/cmake/gz-cmake3/cmake3/FindAVUTIL.cmake
[... cut ...]
TypeError: WPARAM is simple, so must be an int object (got NoneType)
```
repo version
```
PS C:\jazzy\src\gazebo-release\gz_cmake_vendor> git status
On branch jazzy
Your branch is up to date with 'origin/jazzy'.

nothing to commit, working tree clean
PS C:\jazzy\src\gazebo-release\gz_cmake_vendor> git log
commit cf084f5c5e4f55412fd14ccbbdcdad4b1e39264f (HEAD -> jazzy, tag: 0.0.7, origin/jazzy)
Author: Addisu Z. Taddese <addisu@openrobotics.org>
Date:   Thu Apr 25 16:07:36 2024 -0500

    0.0.7
```

clalancette, 2024-05-01:
> I think the 3.X series should work. I'm going to suggest you go ahead and try out the latest (3.3.0), and see if it builds and works. If so, we can update the documentation.

OK, it turns out that there was a bug in CMake that will not allow OpenSSL 3.3 packages to work.  It was fixed at the beginning of April in https://github.com/Kitware/CMake/commit/aa86f487c23a080b381b1b605056f8f394269223 , and there is a new release (3.29.2) that includes it.

So I think what we probably need to do here is to switch to installing OpenSSL 3.3.0 via chocolatey in the installation instructions.  That will also mean we need to update CMake to at least 3.29.2.  And finally, we'll have to update CI so that it does the same thing we are instructing users to do.

I'll get started on making all of that happen, it will take a few hours.

prototriangle, 2024-05-01:
Not sure it's relevant but want to highlight that there's an OpenSSL binary provided at https://www.zlatkovic.com/pub/libxml/64bit/ currently at v1.0.2e if it's preferable to not use a newer version.

clalancette, 2024-05-02:
I just wanted to leave an update here to say we have been working on the issue.  To fix this properly, we are going to have to do a number of upgrades, at least:

* https://github.com/ros/resource_retriever/pull/96 (and a backport to Jazzy)
* https://github.com/gazebosim/gz-cmake/pull/425 (and a backport to Jazzy)
* https://github.com/ros2/ros2_documentation/pull/4343 (and a backport to Jazzy)
* https://github.com/ros-infrastructure/ros2-cookbooks/pull/68
* https://github.com/ros2/ci/pull/778

(I also think it is likely we'll need one or two more PRs)

We'll keep working through this in the next couple of days.

clalancette, 2024-05-13:
All right.  It turned out that due to various other problems, we couldn't actually update to OpenSSL 3.3.0 on Windows (though this is something we are going to look at for post-Jazzy).  Instead, we actually switched both our CI and the instructions to install OpenSSL 1.1.1 from chocolatey, which is exactly what people did to workaround this issue during testing.

With that, I'm going to close this out as done.  Thanks for testing and the patience here, all.  Going ahead and closing this one.

closed 2024-05-13 by clalancette