osrf/ros2_test_cases #1527: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* Build type: Sounce
* Chip set: AMD64
* DDS: ConnextDDS
* OS: Windows

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

srsaito, 2024-05-02:
Wasn't able to install on Windows Server 2022 EC2 instance. Got following errors after completing install instructions:
```
C:\Users\Administrator>call C:\dev\ros2_jazzy\local_setup.bat
"[rti_connext_dds_cmake_module][warning] RTI Connext DDS environment script not found (\resource\scripts\rtisetenv_x64Win64VS2017.bat). RTI Connext DDS will not be available at runtime, unless you already configured PATH manually."

C:\Users\Administrator>ros2 run demo_nodes_cpp talker
[ros2run]: Process exited with failure 3221226505

C:\Users\Administrator>
```
Shall I document in this ticket?

clalancette, 2024-05-02:
The warning for the `RTI Connext` stuff can generally be ignored, that's OK in this case.  The fact that it totally failed is obviously a problem.

First, can you give the commands you used to build with?  I just want to make sure that is OK.

Second, what happens if you do `cd C:\dev\ros2_jazzy\install\Lib\demo_nodes_cpp`, and then run `talker.exe`?  That seems to work here for me, but there may be something else going on with your setup.

srsaito, 2024-05-03:
[build_commands.txt](https://github.com/osrf/ros2_test_cases/files/15195373/build_commands.txt)
1. Attaching transcript of commands I entered for the installation. 
2. The directory path you shared above doesn't exist in my `C:\dev\ros2_jazzy\` directory. That is, there is no `\install\` in that directory.

marcoag, 2024-06-26:
Closing as it's been more than a month of the tutorial party. Thanks for the work here @srsaito.

closed 2024-06-26 by marcoag