osrf/ros2_test_cases #1532: Local workspace overlay

marcoag, 2024-04-30:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Configuration

* Build type: Sounce
* DDS: CycloneDDS
* Chip set: AMD64
* OS: Windows

## Process

    
            
### Local workspace overlay works

mikaelarguedas, 2024-05-06:
Duplicate of https://github.com/osrf/ros2_test_cases/issues/1531 as its RMW independant ?

clalancette, 2024-05-07:
Yeah, probably a duplicate.  I'm going to leave it open for now anyway, but we should probably find a way to omit it next year.

closed 2024-06-26 by marcoag