osrf/ros2_test_cases #1537: RQT

marcoag, 2024-04-30:
This is a sanity check to ensure that the GUI tools aren't fundamentally broken.

## Configuration

* Chip set: AMD64
* Build type: Sounce
* DDS: FastDDS
* OS: RHEL 9

## Links

* [Repo URL](https://github.com/ros-visualization/rqt)

## Process

    
            
### Check RQT
            
        

    
Launch using the `rqt` command.

The rqt window should be generally usable and should be able to populate some of the discovered plugins.

        

    
        
            
#### Actions
            
        
        
```bash
# StdIn - terminal 1
ros2 launch dummy_robot_bringup dummy_robot_bringup_launch.py
```
                
```bash
# StdIn - terminal 2
rqt
```
                
In the RQT window, select 'Plugins > Visualization > Plot' in the dropdown menu. In the window that opens, set the Topic to `/joint_states/position[0]`. You should see a Sine wave.
                
With 'Plugins > Introspection > Node Graph' you should see a network of the nodes and topics that are being published and subscribed by the dummy robot.
                
With 'Plugins > Logging > Bag' you should be able record a ROS Bag file.
                
With 'Plugins > Topic > Topic Monitor' you should be able to check the `/joint_states` topic and view messages streaming in.
                
    

    
        
#### Expected Result
        
        
Rqt runs smoothly and the plugins work as expected.

hwoithe, 2024-05-03:
I created a Rocky Linux 9.3 container and followed the installation instructions [here](https://docs.ros.org/en/jazzy/Installation/Alternatives/RHEL-Development-Setup.html).
I did not compile the sources with Clang.

1. rqt comes up successfully:
[image]

2. Selecting Plugins->Visualization->Plot in the RQT window output the following error:
```
PluginManager._load_plugin() could not load plugin "rqt_plot/Plot":
Traceback (most recent call last):
  File "/home/hwoithe/tmp/jazzy/install/qt_gui/lib/python3.9/site-packages/qt_gui/plugin_handler.py", line 102, in load
    self._load()
  File "/home/hwoithe/tmp/jazzy/install/qt_gui/lib/python3.9/site-packages/qt_gui/plugin_handler_direct.py", line 55, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/home/hwoithe/tmp/jazzy/install/qt_gui/lib/python3.9/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/home/hwoithe/tmp/jazzy/install/qt_gui/lib/python3.9/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/home/hwoithe/tmp/jazzy/build/rqt_gui_py/src/rqt_gui_py/ros_py_plugin_provider.py", line 69, in load
    return super(RosPyPluginProvider, self).load(plugin_id, ros_plugin_context)
  File "/home/hwoithe/tmp/jazzy/install/qt_gui/lib/python3.9/site-packages/qt_gui/composite_plugin_provider.py", line 72, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/home/hwoithe/tmp/jazzy/build/rqt_gui/src/rqt_gui/ros_plugin_provider.py", line 107, in load
    return class_ref(plugin_context)
  File "/home/hwoithe/tmp/jazzy/build/rqt_plot/src/rqt_plot/plot.py", line 53, in __init__
    self._data_plot = DataPlot(self._widget)
  File "/home/hwoithe/tmp/jazzy/build/rqt_plot/src/rqt_plot/data_plot/__init__.py", line 155, in __init__
    raise RuntimeError(
RuntimeError: No usable plot type found. Install at least one of: PyQtGraph, MatPlotLib (at least 1.4.0) or Python-Qwt5.
```

3. Selecting Plugins->Introspection->Node Graph produces:
[image]

4. Selecting Plugins->Logging->Bag outputs the following error:
```
RosPluginProvider.load(rqt_bag_plugins/BagImagePlugin) exception raised in __builtin__.__import__(rqt_bag_plugins.image_plugin, [ImagePlugin]):
Traceback (most recent call last):
  File "/home/hwoithe/tmp/jazzy/build/rqt_gui/src/rqt_gui/ros_plugin_provider.py", line 80, in load
    module = __builtin__.__import__(
  File "/home/hwoithe/tmp/jazzy/build/rqt_bag_plugins/src/rqt_bag_plugins/image_plugin.py", line 35, in <module>
    from .image_timeline_renderer import ImageTimelineRenderer
  File "/home/hwoithe/tmp/jazzy/build/rqt_bag_plugins/src/rqt_bag_plugins/image_timeline_renderer.py", line 44, in <module>
    from PIL.ImageQt import ImageQt
ModuleNotFoundError: No module named 'PIL.ImageQt'

rqt_bag.TimelineFrame.load_plugins() failed to load plugin "rqt_bag_plugins/BagImagePlugin":
No module named 'PIL.ImageQt'
[INFO] [1714766585.817895624] [rqt_bag.BagWidget]: Recording to /tmp/2024-05-03-16-02-58.
Error opening bag for recording [/tmp/2024-05-03-16-02-58]: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. rosbag2_py._storage.TopicMetadata(id: int, name: str, type: str, serialization_format: str, offered_qos_profiles: List[rosbag2_py._storage.QoS] = [], type_description_hash: str = '')

Invoked with: kwargs: name='/joint_states', type='sensor_msgs/msg/JointState', serialization_format='cdr', offered_qos_profiles='- history: 3\n  depth: 0\n  reliability: 1\n  durability: 2\n  lifespan:\n    sec: 9223372036\n    nsec: 854775807\n  deadline:\n    sec: 9223372036\n    nsec: 854775807\n  liveliness: 1\n  liveliness_lease_duration:\n    sec: 9223372036\n    nsec: 854775807\n  avoid_ros_namespace_conventions: false\n'
```

5. Selecting Plugins->Topic->Topic Monitor produces:
[image]

Here is my system's information:
```
$ cat /etc/redhat-release
Rocky Linux release 9.3 (Blue Onyx)

$ glxinfo | grep -i renderer
OpenGL renderer string: NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2
```

marcoag, 2024-06-26:
Closing as we are over a month after tutorial party ended. We'll keep track of this issues as it might have some follow up inolved, thanks for the work here @hwoithe!

closed 2024-06-26 by marcoag