osrf/ros2_test_cases #1538: Installation instructions

marcoag, 2024-04-30:
Follow the official installation instructions for the platform. If instructions have not been published for the platform, use the instructions for Rolling Ridley.

For source build platforms, this includes the installation of dependencies and building the packages.

## Configuration

* Chip set: AMD64
* Build type: Sounce
* DDS: FastDDS
* OS: RHEL 9

## Links

* [Installation instructions](https://docs.ros.org/en/jazzy/Installation.html)

## Process

    
            
### Install works

hwoithe, 2024-05-03:
I created a Rocky Linux 9.3 container and followed the installation instructions[ here](https://docs.ros.org/en/jazzy/Installation/Alternatives/RHEL-Development-Setup.html).

I did receive two warning messages while compiling:
```
$ gcc --version
gcc (GCC) 11.4.1 20230605 (Red Hat 11.4.1-2)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ g++ --version
g++ (GCC) 11.4.1 20230605 (Red Hat 11.4.1-2)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ cd ~/tmp/jazzy
$ rm -fr build install log
$ colcon build --symlink-install
...
[165.812s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'rosbag2_performance_benchmarking_msgs' because it has no 'install' target 
...
[422.688s] colcon.colcon_cmake.task.cmake.build WARNING Could not run installation step for package 'rosbag2_performance_benchmarking' because it has no 'install' target
...

```

I was able to run the demo node talker and listener:
[image]

Compiling with Clang was not as successful:
```
$ clang --version
clang version 16.0.6 (Red Hat 16.0.6-1.el9)
Target: x86_64-redhat-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

$ clang++ --version
clang version 16.0.6 (Red Hat 16.0.6-1.el9)
Target: x86_64-redhat-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

$ export CC=clang
$ export CXX=clang++
$ cd ~/tmp/jazzy
$ rm -fr build install log
$ colcon build --symlink-install
...
Starting >>> iceoryx_hoofs
...
--- stderr: iceoryx_hoofs                                                                                                                                      
/opt/rh/gcc-toolset-13/root/usr/lib/gcc/x86_64-redhat-linux/13/../../../../bin/ld: cannot find -latomic: No such file or directory
clang-16: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [CMakeFiles/iceoryx_hoofs.dir/build.make:658: libiceoryx_hoofs.so] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:100: CMakeFiles/iceoryx_hoofs.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
---
Failed   <<< iceoryx_hoofs [3.12s, exited with code 2]
...

$ dnf list --installed | grep -i atomic
boost-atomic.x86_64                                 1.75.0-8.el9                        @appstream
libatomic.x86_64                                    11.4.1-2.1.el9                      @baseos
```

I then decided to add a few exceptions:
```
$ rm -fr build install log
$ colcon build --symlink-install --packages-skip iceoryx_hoofs iceoryx_posh iceoryx_binding_c iceoryx_introspection cyclonedds rmw_cyclonedds_cpp
...
--- stderr: lttngpy                                                                                                                                                              
In file included from /home/hwoithe/tmp/jazzy/src/ros2/ros2_tracing/lttngpy/src/lttngpy/_lttngpy_pybind11.cpp:20:
In file included from /usr/include/pybind11/pybind11.h:45:
/usr/include/pybind11/attr.h:199:5: error: no type named 'uint16_t' in namespace 'std'; did you mean simply 'uint16_t'?
    std::uint16_t nargs;
    ^~~~~
/usr/include/bits/stdint-uintn.h:25:20: note: 'uint16_t' declared here
typedef __uint16_t uint16_t;
                   ^
In file included from /home/hwoithe/tmp/jazzy/src/ros2/ros2_tracing/lttngpy/src/lttngpy/_lttngpy_pybind11.cpp:20:
In file included from /usr/include/pybind11/pybind11.h:45:
/usr/include/pybind11/attr.h:202:5: error: no type named 'uint16_t' in namespace 'std'; did you mean simply 'uint16_t'?
    std::uint16_t nargs_kw_only = 0;
    ^~~~~
/usr/include/bits/stdint-uintn.h:25:20: note: 'uint16_t' declared here
typedef __uint16_t uint16_t;
...
```

Adding ```lttngpy``` and ```tracetools_trace``` to the skip list lets the build proceed a bit further.
We eventually, however, have to skip core packages such as ```rmw_implementation```.

Here is my system's information:
```
$ cat /etc/redhat-release
Rocky Linux release 9.3 (Blue Onyx)

$ glxinfo | grep -i renderer
OpenGL renderer string: NVIDIA GeForce RTX 3060 Ti/PCIe/SSE2
```

clalancette, 2024-05-06:
Thanks for testing!

I think we can ignore the `clang` failures; while that is good to know, that is not one of our primary targets here (and we do have clang building with Ubuntu).

The other warnings are unique to RHEL, and not really a problem.

So with all of that, I'm going to go ahead and close this.  Thanks for testing!

closed 2024-05-06 by clalancette