osrf/ros2_test_cases #1539: Local workspace overlay

marcoag, 2024-04-30:
Build a package in a local workspace which depends on packages supplied for the platform.

Run the tests for the package - all of them should pass.

For source platforms, this means building the candidate package in a separate workspace from the one used in the installation instructions.

TODO: Create a "dummy" package that ensures that we can target other packages using CMake, link against them, and run executables (tests) that are linked against libraries in the main workspace.

## Configuration

* Chip set: AMD64
* Build type: Sounce
* DDS: FastDDS
* OS: RHEL 9

## Process

    
            
### Local workspace overlay works

clalancette, 2024-05-01:
Need a small fix to the RHEL documentation: https://github.com/ros2/ros2_documentation/pull/4340

clalancette, 2024-05-01:
Other than the documentation update (which I've already merged), this seemed to work for me, so closing.

closed 2024-05-01 by clalancette