osrf/ros2_test_cases #1549: /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash: No such file or directory

clemefr, 2024-05-04:
in Setup colcon tab completion tutorial
echo "source /usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash" >> ~/.bashrc

/usr/share/colcon_argcomplete/hook/colcon-argcomplete.bash: No such file or directory

christophebedard, 2024-05-13:
I believe you're referring to this section: https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#setup-colcon-tab-completion

How did you install `colcon`? It depends on how you installed `colcon`, which is mentioned in the docs:

> Depending on the way you installed colcon and where your workspace is, the instructions above may vary, please refer to [the documentation](https://colcon.readthedocs.io/en/released/user/installation.html) for more details.

It should probably directly link to the right section: https://colcon.readthedocs.io/en/released/user/installation.html#enable-completion. If you've installed it using `pip`, then the file would be under `/usr/local/share/colcon_argcomplete/hook`.

clalancette, 2024-05-13:
> It should probably directly link to the right section: https://colcon.readthedocs.io/en/released/user/installation.html#enable-completion. If you've installed it using `pip`, then the file would be under `/usr/local/share/colcon_argcomplete/hook`.

Oh, great call.  Thanks @christophebedard .  In point of fact, all of that information is *already* on the page, we should just remove the `echo` commands to simplify it.  I'll open a PR to do that.

closed 2024-05-13 by clalancette

clemefr, 2024-05-14:
> I believe you're referring to this section: https://docs.ros.org/en/jazzy/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#setup-colcon-tab-completion
>
> How did you install `colcon`? It [...]

clemefr, 2024-05-14:
I install following the tutorial from the beginning