osrf/ros2_test_cases #1555: ros2cli on a local machine - WLS2 Ubuntu 24.04 on Win11

knmcguire, 2024-05-07:
Test the basic functionality of the ros2cli package family

**Configuration**
Chip set: AMD64
Build type: Debian
DDS: FastDDS
OS: WLS2 Ubuntu Noble on Win 11

**Tests**
Similar to those listed in https://github.com/osrf/ros2_test_cases/issues/1429

knmcguire, 2024-05-07:
- [ ] Step 1 - Help text
Works fine
- [ ] Step 2 - Tab completion 
Works fine
- [ ] Step 3 - Node discovery

Terminal 1:
```
$ ros2 launch demo_nodes_cpp talker_listener_launch.py
[INFO] [launch]: All log files can be found below /home/knmcguire/.ros/log/2024-05-07-16-46-06-557771-kim-legion-1118
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [1121]
[INFO] [listener-2]: process started with pid [1122]
[talker-1] [INFO] [1715093167.657423485] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1715093167.657955546] [listener]: I heard: [Hello World: 1]
[talker-1] [INFO] [1715093168.649167895] [talker]: Publishing: 'Hello World: 2'
[listener-2] [INFO] [1715093168.649786089] [listener]: I heard: [Hello World: 2]
[talker-1] [INFO] [1715093169.670622026] [talker]: Publishing: 'Hello World: 3'
[listener-2] [INFO] [1715093169.671357586] [listener]: I heard: [Hello World: 3]
```
Terminal 2:
```
$ source /opt/ros/jazzy/setup.bash
$ ros2 node list
```

Interesting nothing to be found... eventhough rqt does show nodes

- [ ] Step 4 - Topic discovery

Terminal 2: 
```
$ ros2 topic list
/chatter
/parameter_events
/rosout
```

- [ ] Step 5  - Service discovery

terminal 1:
```
$ ros2 run demo_nodes_cpp add_two_ints_server
```

terminal  2:
```
$ ros2 service list
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/get_type_description
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically
```

- [ ] Step 6 - Parameter discovery

Terminal 1:
```
$ ros2 run turtlesim turtlesim_node
[INFO] [1715093758.669962158] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1715093758.679863918] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
```

Terminal 2:
```
$ ros2 param list
```
Strangely nothing here either, but rqt dynamic configuration tab does work without issue.

knmcguire, 2024-05-07:
The ros2cli help viewing, tab completion, service and topic listing works but the discovery of nodes/params **does not seem to work.** 

Interestingly the last two do work in rqt...

knmcguire, 2024-05-08:
Hmmm restarting my computer with everything seems to work again! Perhaps  there were too many docker containers open or such, might have been a timing issue? I did had one wsl of 22.04 open, and a jazzy docker on powershell, so who knows what kind of things that does to the network complexity.

It was weird that only those 2 discovery functions seems to fail at that time... but I'll keep my eye open and see if it happens again.

Step 3 - Node discovery
```
$ ros2 node list
/listener
/talker

```
Step 6 - Parameter discovery
```
$ ros2 param list
/turtlesim:
  background_b
  background_g
  background_r
  holonomic
  qos_overrides./parameter_events.publisher.depth
  qos_overrides./parameter_events.publisher.durability
  qos_overrides./parameter_events.publisher.history
  qos_overrides./parameter_events.publisher.reliability
  start_type_description_service
  use_sim_time
```

So all good for now? Or do you still want me to try cyclonedds just out of curiousity @clalancette ?

clalancette, 2024-05-08:
> So all good for now? Or do you still want me to try cyclonedds just out of curiousity @clalancette ?

It would be nice to know if it worked, but totally up to you.  This is "bonus" testing :).

knmcguire, 2024-05-08:
**step 3 & 4 with cyclonedds**

T1
```
$ source /opt/ros/jazzy/setup.bash
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 launch demo_nodes_cpp talker_listener_launch.py
[INFO] [launch]: All log files can be found below.../.ros/log/2024-05-08-17-13-55-641306-kim-legion-750
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [talker-1]: process started with pid [754]
[INFO] [listener-2]: process started with pid [755]
[talker-1] [WARN] [1715181235.755568224] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'ros_discovery_info' with type 'rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_' from USER_DATA '(null)'.
[talker-1] [WARN] [1715181235.757944667] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'ros_discovery_info' with type 'rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_' from USER_DATA '(null)'.
[talker-1] [WARN] [1715181235.757951089] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/rosout' with type 'rcl_interfaces::msg::dds_::Log_' from USER_DATA '(null)'.
[talker-1] [WARN] [1715181235.757955277] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/parameter_events' with type 'rcl_interfaces::msg::dds_::ParameterEvent_' from USER_DATA '(null)'.
[talker-1] [INFO] [1715181236.772511524] [talker]: Publishing: 'Hello World: 1'
[listener-2] [INFO] [1715181236.772850456] [listener]: I heard: [Hello World: 1]
```

T2
```
$ source /opt/ros/jazzy/setup.bash
$ export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
$ ros2 node list
/listener
/talker
$ ros2 topic list
/chatter
/parameter_events
/rosout
```

**Step 5 with cyclonedds**

T1
```
$ ros2 run demo_nodes_cpp add_two_ints_server
```
T2
```
$ ros2 service list
/add_two_ints
/add_two_ints_server/describe_parameters
/add_two_ints_server/get_parameter_types
/add_two_ints_server/get_parameters
/add_two_ints_server/get_type_description
/add_two_ints_server/list_parameters
/add_two_ints_server/set_parameters
/add_two_ints_server/set_parameters_atomically
```

**step 6 with cyclonedds**
T1
```
$ ros2 run turtlesim turtlesim_node
[INFO] [1715181504.281245046] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1715181504.284943605] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
```

T2
```
$ ros2 run turtlesim turtlesim_node
[INFO] [1715181504.281245046] [turtlesim]: Starting turtlesim with node name /turtlesim
[INFO] [1715181504.284943605] [turtlesim]: Spawning turtle [turtle1] at x=[5.544445], y=[5.544445], theta=[0.000000]
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen

```

Seems to work okayish, with some warnings at step1 and 2

clalancette, 2024-05-08:
> Seems to work okayish, with some warnings at step1 and 2

Hm, that should only ever happen if you are mixing pre-Iron and post-Iron publishers/subscribers on the same network.  Is that the case in your setup?

knmcguire, 2024-05-08:
ahhh yes makes sense. I'm currently working on humble on the side on a different project on the 22.04 wsl2. I've just changed the ROS_DOMAIN_ID for the 24.04 wsl2 for both terminals and the warning has disappeared.

knmcguire, 2024-05-17:
Ah I see this is still open, but last week I confirmed that it is all working with cyclone dds as well.

I've just confirmed again with a restart and without setting a ROS_DOMAIN_ID (since no other ROS distros active in the same network), and all the tests worked fine.

clalancette, 2024-05-17:
Sounds good, thanks for testing again!  Closing this out.

closed 2024-05-17 by clalancette