osrf/ros2_test_cases #1558: ros2 launch urdf_tutorial display.launch.py error

clemefr, 2024-05-08:
pi@pc-ubun90:~/Desktop$ ros2 launch urdf_tutorial display.launch.py model:=urdf/01-myfirst.urdf
[INFO] [launch]: All log files can be found below /home/pi/.ros/log/2024-05-08-17-39-33-340479-pc-ubun90-12367
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [12371]
[INFO] [joint_state_publisher_gui-2]: process started with pid [12372]
[INFO] [rviz2-3]: process started with pid [12373]
[robot_state_publisher-1] [INFO] [1715182773.550040036] [robot_state_publisher]: Robot initialized
[rviz2-3] QSocketNotifier: Can only be used with threads started with QThread
[rviz2-3] [INFO] [1715182773.804959508] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1715182773.805032972] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [ERROR] [1715182773.825862868] [rviz2]: rviz::RenderSystem: error creating render window: RenderingAPIException: Invalid parentWindowHandle (wrong server or screen) in GLXWindow::create at ./.obj-x86_64-linux-gnu/ogre_vendor-prefix/src/ogre_vendor/RenderSystems/GLSupport/src/GLX/OgreGLXWindow.cpp (line 246)
...

clalancette, 2024-05-08:
This usually happens when you can't properly create the rviz window.  In turn, that either means that you don't have X11 forwarded properly, or that you are using wayland.  Can you try:

1.  `$ glxgears`?  If that works, then at least you know you can create windows
2.  If 1 is successful, then follow the advice in http://docs.ros.org/en/jazzy/How-To-Guides/Installation-Troubleshooting.html#cannot-start-rviz2 to try to make it start.

clemefr, 2024-05-11:
1 successful
2 export QT_QPA_PLATFORM=xcb rviz2 successful

Thanks a lot

clalancette, 2024-05-12:
Thanks for testing.

closed 2024-05-12 by clalancette