osrf/ros2_test_cases #1559: Executables in gz vendor packages

knmcguire, 2024-05-14:
Copied from https://github.com/osrf/ros2_test_cases/issues/1274

You will need to install the desktop-full or simulation variants before running this test.
## Configuration

- DDS: FastDDS
-  Build type: Debian
- OS: Ubuntu Noble 24.04
- Chip set: ARM64

## Process
Check gz sim -v4 shapes.sdf

```
# StdIn - terminal 1
gz sim -v4 shapes.sdf
```

knmcguire, 2024-05-14:
By request of @azeey I've installed desktop full on a raspberry pi with Ubuntu 24.04 installed and tried out gazebo. The gui does start up but it crashes and I get a system error that ruby has crashed. 

I've attached a text file with the crash here:

[gazebo_crash_raspberrypi_log.txt](https://github.com/osrf/ros2_test_cases/files/15312106/gazebo_crash_raspberrypi_log.txt)

clalancette, 2024-05-14:
> I've attached a text file with the crash here:

Does rviz2 work on this setup?  Because it looks like Gazebo failed to create the rendering window.

knmcguire, 2024-05-14:
Ah no it doesn't indeed. Let me try that fix from earlier

knmcguire, 2024-05-14:
No dice!

 `QT_QPA_PLATFORM=xcb gz sim -v4 shapes.sdf` still fails to open the rendering window.

Rviz2 still works with `QT_QPA_PLATFORM=xcb rviz2`

I'll debug it further. Seeing some opengl related errors  as well

knmcguire, 2024-05-14:
So, it seems that Gazebo needs at least OpenGl 3.3, and raspberry pi ubuntu only support up to 3.1
```
$ glxinfo | grep -i 'opengl version'
OpenGL version string: 3.1 Mesa 24.0.5-1ubuntu1
```

When I look at t[he gazebo troubleshooter,](https://gazebosim.org/docs/harmonic/troubleshooting#unable-to-create-the-rendering-window) it says I can check the ogre2.log:

[ogre2crash.txt](https://github.com/osrf/ros2_test_cases/files/15312827/ogre2crash.txt)
```
20:48:06: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at ./.obj-aarch64-linux-gnu/gz_ogre_next_vendor-prefix/src/gz_ogre_next_vendor/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3434)
```

knmcguire, 2024-05-14:
fwew, Had to test out several combinations of all the tips of the troubleshooter, but this was the winner:
```
LIBGL_ALWAYS_SOFTWARE=1 QT_QPA_PLATFORM=xcb gz sim -v 4 shapes.sdf
```

See the screenshot here
[image]

Poor RP5 had to work hard on that and the sim could only run 60% speed with difficulty, but it runs!

ahcorde, 2024-05-15:
Thank you for testing

closed 2024-05-15 by ahcorde